EdgeTX/edgetx #4979: Brightness adjustment is disabled in 2.10 RC3

udayan23, 2024-05-07:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When we set function in 2.9.4 for brightness adjustment on S1 or S2 pot, it is now disabled in 2.10 RC3. It is not there in RC4 also.

### Expected Behavior

Please activate the brightness adjustment.

### Steps To Reproduce

1. developer knows the steps

### Version

Other (Please specify below)

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

Rest everything is best.

pfeerick, 2024-05-07:
Have you gone into the special function and enabled it again? If you have just updated from 2.9.x it is probably disabled. There was mention of 

> When updating, the following special and/or global functions will need to be re-enabled: Instant Trim, BG Music Pause, Vario, Screenshot, Racing Mode, Disable Touch, and Audio Amp Off. cf. https://github.com/EdgeTX/edgetx/pull/4498

in the release notes under "Known Limitations and Issues" but perhaps it extends to Volume and Backlight special functions also. Or might just rephrase that to generally check there if a function has stopped working.

udayan23, 2024-05-07:
> Have you gone into the special/general function and enabled it again? If

Sir

I have tried all the ways possible. I think it is permanently disabled from
the program.

Regards

pfeerick, 2024-05-07:
It is setup something like this, and enabled? 
[image]

If you're still having trouble and it is configured properly, can you upload a copy of your RADIO and MODELS folder (as a zip file) -  you'll need to upload it on Github as email attachment don't come through.

ParkerEde, 2024-05-07:
with my two RM TX16S MKII and 2.10 RC3 and RC4 it works exactly as it should with Peter's screenshot.

tommyseus, 2024-05-20:
My issue was that the slider for ON was on the lowest value. 

Radio Settings - Radio Setup - Backlight