EdgeTX/edgetx #4983: Companion Telemetry Simulator does not support ELRS/CRSF Telemetry

icebreaker-ch, 2024-05-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Simulator

### Current Behavior

I am using a Radiomaster 2.4 GHz ELRS-Module and a Radiomaster ER8GV Receiver having a Vario).
I was not able, to simulate telemetry values for this configuration.
Screenshots of the Telemetry page of the Transmitter and the Telemetry Simulator screen are  attached in this report. It seems, that the Telemtry Simulator is limited to FrSky Telemetr values. ELRS uses the Crossfire protocol and all telemetry values appear as "Custom" on the Radio.
[image]
[image]

### Expected Behavior

I would like to have possibility, to configure custom sensors for the Telemetry Simulator or some kind od automatic recognition of the telemetry values.

### Steps To Reproduce

- Configure a radio for an ELRS receiver
- Discover all sensors
- Save configuration to Companion
- Start a Simulation
- Modify Values (e.g. Vario Alt)
Sensors do not match the ones in the radio

### Version

2.9.4

### Transmitter

FrSky X12

### Operating System (OS)

Linux

### OS Version

LMDE 6 Faye

elecpower, 2024-05-08:
also refer #3047

3djc, 2024-05-08:
Same can be said to any non FRSKY telemetry (Spektrum, HoTT,....)

rotorman, 2024-08-02:
https://edgetx.org/contest_crsf_telem_simulation/

PR #5410 (2024-08-20): feat(cpn): support more than just FrSky S.Port telemetry simulation
Fixes #4983
Fixes #2156 

Summary of changes:

1. Replace the current telemetry sensor UI dropdowns to choose a telemetry source for both the internal and external module.
2. Refactor the current FrSky S.port telemetry implementation to allow it to have separate telemetry for both internal and external modules at the same time
3. Update the simulator interface to accept different telemetry protocols
4. Implement a CRSF telemetry source and associated UI
5. Rejigger the way the log replay controller works so that it can work wi [... cut]
closed 2024-08-20 by pfeerick