EdgeTX/edgetx #4984: EdgeTX flasher bricks the receiver if the bootloader is being updated.

blopa1961, 2024-05-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

A patch for Archer Plus R6 | GR6 | R8 | GR8 | SR8 which "Fixed the issue of CH1/CH2 outputting an unexpected pulse width once powering the receiver." was posted recently by FrSky.
The patch comes in 2 files which must be flashed consecutively.
The first file "nx_Patch.frsk" contains an updated bootloader (encrypted) and the second "nx_FixTool.frsk" is the bootloader updater (also encrypted).

Here are the included instructions:
```
[Notice]: Please ensure the OTA function is unworking when using this tool.
Step 1: Put the .frsk files into the SD card of the radio. 
Step 2: Connect the receiver to the radio by S.Port. 
Step 3: Update the Patch.frsk file to the receiver by "Flash External Device".
Step 4: Update the FixTool.frsk file to the receiver by "Flash External Device".
Step 5: Disconnect the receiver from the radio 
Step 6: Repower the receiver with a battery/power source, the LED_Red and LED_Green will turn ON simultaneously, then the LED_Red gets off and LED_Green starts flashing.
Step 7: Flash the regular firmware such as AP_R6_GR6_R6FB_GR6FB_1.0.10.frsk or AP_8CH_1.0.10.frsk to the receiver.
Step 8: Done and enjoy your flight.
```

It seems that either the EdgeTX flasher clears the whole receiver's flash before flashing a new one, or just overwrites the preloaded/encrypted/updated bootloader patch when the nx_FixTool is flashed.

FrSky currently pulled this patch, it's no longer available in their site.

### Expected Behavior

OpenTX's flasher works ok and EdgeTX's changes in the flasher should be rolled back

### Steps To Reproduce

If you reproduce this problem you will brick your receiver!

### Version

Nightly (Please give date/commit below)

### Transmitter

FrSky X9 Lite / Lite S

### Operating System (OS)

Windows

### Anything else?

firmware: nightly e3f29148
2024-04-03

blopa1961, 2024-05-14:
I was the one who helped FrSky pinpoint the problem, and now that the culprit was identified as EdgeTX they are refusing to replace my bricked receivers free of charge (even when the second receiver I bricked was following their instructions with an updated patch).
They sent me a video applying the patch with OpenTX without problems, so, it IS the EdgeTX flasher what causes the bricking. They removed the patch from their website so that we, EdgeTX users, stop bricking receivers.
Granted that this particular patch was vey poorly engineered, if it were a single file this would never have happened.

ParkerEde, 2024-05-14:
Oh, that sounds bitter. Is it no longer possible to revive the Rx with the STM32Cube Programmer?

blopa1961, 2024-05-14:
These are ACCESS receivers; the chip (which is sanded so as not to be able to identify it) is probably a C25xx, which runs some kind of 8051 code. All ACCESS firmware is encrypted, FrSky wants ACCESS to remain proprietary (since D16 was copied by other brands). So, these are nice bricks but too light for a paperweight... A pity we can not eve flash some open source firmware.