EdgeTX/edgetx #4985: After updating to EdgeTX 2.9.4 only partial telemetry is discovered

Avi-Levy1, 2024-05-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating my radiomaster TX16s to the latest version of EdgeTX (2.9.4) only five sensors are discovered.
Previously, I had  EdgeTX 2.6 and 16 sensors were discovered.
 
Now, only 5 sensors are discovered, namely: RSSI, TRSS, TQly, RxBt, GPS
Previously  16 sensors were discovered, namely: RSSI, TRSS, TQly, RxBt, Fuel, CURR, Hdg, Alt, GSpd, VFAS, Vspd, GAlt, IMUt, ARM, Thr, GPS
 
Using jumper r8 RX connected to Pixhawk - binding - Frsky X - D16

### Expected Behavior

discover all the sensors

### Steps To Reproduce

deleted all the telemetry and press discover new

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

blopa1961, 2024-05-09:
This might not be a bug...
In pixhawk you can choose two FrSky telemetry protocols, either 04 or 10.
If you chose protocol 10, this is a **_passthrough_** protocol which is used by iNav LUA directly; in the radio telemetry you will only see the pixhawk's GPS.
If you intend to run the iNav LUA, it should already be working ok with full telemetry.
If you want to use the individual sensors you have to change the pixhawk protocol to FrSky (04), but iNav will not work in this mode.
I hope this helps.

Avi-Levy1, 2024-05-09:
Thanks
I am using yaapu.
In Edge 2.6 I have used protocol 10 and it worked good.
protocol 4, is for Frsky D s.port only, while I am, using the telemetry port and not the serial port.
I have try to put 4, but the results are the same

blopa1961, 2024-05-09:
I'm using yaapu with pixhawk telemetry protocol 10 with an R-XSR receiver in ACCESS mode using S.PORT via a hardware inverter. It works ok, yaapu decodes the full telemetry.
Since this is the passthrough protocol, I only get the receiver's telemetry plus the pixhawk GPS in the telemetry menu, which looks very much like the 5 sensors you get (4 from the receiver and just GPS from the pixhawk); that's why I guessed it was the same problem.
My X9 Lite S is running 2.11.0 nightly.

Avi-Levy1, 2024-05-09:
I'm using yaapu with pixhawk telemetry protocol 10 with jumper R8
ACCST receiver. Connected to the telemetry port (TX RX).
It worked well with Edge2.6 while I could see all the sensors as described.
The change I made was upgrading the Eage to 2.9.4

Form your reply, I understand that the yaapu script should talk with the Pixhawk and get the telemetry. Did I understand you correctly?

Do I need to change the yaapu script in order to solve it?

blopa1961, 2024-05-09:
Yes. When you use protocol 10, the radio does not decode the sensors (it only decodes GPS), it passes the data directly to the yaapu LUA, thus, yaapu should work ok even if you don't "see" the telemetry in the radio's telemetry menu.
If you change the protocol to 4 you will see all sensors in the menu but yaapu will not work.
You can test it, change the protocol to 4 in the pixhawk, cycle pixhawk power and discover new sensors in you radio.
Then go back to protocol 10, yaapu should work ok even if the sensors are not listed in the radio.

Avi-Levy1, 2024-05-11:
Many thanks.
Your procedure solved the problem and yaapu is now working.

[cid:storage_emulated_0_Android_data_com_samsung_android_email_provider_files__EmailTempImage_1_TempSignature_signature_image002_0_jpg_1695810798506]Prof. Avi Levy,
Lady Davis Chair in Mechanical Engineering
          | Faculty of Eng. Sci.
          | Department of Mech. Eng.
Email: ***@***.******@***.***>
Tel: +972-8-6477092 |  Cell: +972-54-6775517<tel:+972-54-6775517>
Please don't print this email unless you really need to
[cid:storage_emulated_0_Android_data_com_samsung_android_email_provider_files__EmailTempImage_1_TempSignature_signature_image003_0_png_1695810798506]

-------- הודעה מקורית --------
מאת: blopa1961 ***@***.***>
תאריך: 9.5.2024 20:49 (GMT+02:00)
אל: EdgeTX/edgetx ***@***.***>
עותק: Avi Levy ***@***.***>, Author ***@***.***>
נושא: Re: [EdgeTX/edgetx] After updating to EdgeTX 2.9.4 only partial telemetry is discovered (Issue #4985)

Yes. When you use protocol 10, the radio does not decode the sensors (it only decodes GPS), it passes the data directly to the yaapu LUA, thus, yaapu should work ok even if you don't "see" the telemetry in the radio's telemetry menu.
If you change the protocol to 4 you will see all sensors in the menu but yaapu will not work.
You can test it, change the protocol to 4 in the pixhawk, cycle pixhawk power and discover new sensors in you radio.
Then go back to protocol 10, yaapu should work ok even if the sensors are not listed in the radio.

blopa1961, 2024-05-11:
Great!
Please **close** this issue.

closed 2024-05-11 by Avi-Levy1