EdgeTX/edgetx #4989: Improve customisable switches with automatic "source "

3djc, 2024-05-11:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently, customisable switches are only switches, and if you want say have it work like a 6POS, it does create a lot of work

### Describe the solution you'd like

Creation of 3 sources (GR1 to GR3) that match the customisable switches
Those source will be managed by the system without user intervention.
They will range from -100 to 100 with as many steps as there are switches in the group. If no switches in group, then 0

### Describe alternatives you've considered

Use SF like today

philmoz, 2024-05-11:
Sounds good.

Although if the group is set to always on then you would get one less step as you can't turn off all the switches in the group.

pfeerick, 2024-05-11:
That's what I'm hoping will happen here... in using this I would want this to simply replace a 6POS properly, so only 6 positions, not 7! :rofl:

3djc, 2024-05-11:
> That's what I'm hoping will happen here... in using this I would want this to simply replace a 6POS properly, so only 6 positions, not 7! 🤣

yes, simply declare group1 always on, and remember also to set your start position either to remember or decide and initial position

closed 2024-06-07 by pfeerick

CapnBry, 2024-06-11:
You magnificent son of a bic. Just got the T15 and thought, man this functional switch groups 6POS is so amazing and so close to perfect but all I need is a GRPx source that behaves **exactly** like you've described. I was browsing the code here trying to figure out where it would need to go and was confused how it already seemed to exist, then the `git blame` lead me here.

A-mazing, everyone is going to be begging for more handsets with hardware that supports this.

pfeerick, 2024-06-11:
> A-mazing, everyone is going to be begging for more handsets with hardware that supports this.

1000% agree... this was really the missing piece since customisation switches were first added... i.e. how to make them make them act like a traditional 6POS without a gazillion mixer lines, logical switches, etc, if you *don't* want the extra functionality offered. It will be disappointing if future radios designs that have a 6POS button implementation don't come with customisation switches as standard, now that this pain point is gone - and in point of fact is far superior to that since you could quite easily have two 3POS, three 2POS switches just by grouping them how you want them!

3djc, 2024-06-12:
You can even have two 4POS since groups don't have to be always on

3djc, 2024-06-12:
Next step is user controlled led color when using ws2812, but it is a UI chalenge :)

PR #5016 (2024-06-07): feat: customisable switches generated sources
This add 3 sources (GR1 to GR3) on radio that  have customisable switches as described in https://github.com/EdgeTX/edgetx/issues/4989

This PR is best when used in conjunction with #4994 (which prevents user misconfiguration)

This fixes #4989 

@elecpower: I've done my  best for companion support, but please check thoroughly.