EdgeTX/edgetx #4990: Quirks in behaviour of customisable function switches.

philmoz, 2024-05-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

There are some oddities in the way the customisable behave (tested on T20V2).

My understanding is that a switch group should follow the following rules.
- only zero or one of the switches in the group can be on at any given time
- if the group is set to always on, then one of the switches must be on

- When a switch group is changed to always on, the radio does not enforce the rules so the group could have 0 or more than one switch turned on depending on the current state of the switches.
- When a switch is added to a group, the rules are not enforced so multiple switches can be turned on.
- If more than one switch on a group has the startup state set to UP or DOWN then the radio will start with all the affected switches in the group turned on - again breaking the only one rule for a group.
- If a switch is removed from an always on group the group could be left with no switch turned on.

Other concerns, with the B&W UI:
- If the switch type is set to None then the group and always on checkbox should be hidden.
- If the switch is part of a group then the switch type 'Toggle' should not be available (the switch will not work as a toggle when in a group).

### Expected Behavior

When editing the custom function switches the rules should always be enforced to prevent invalid states being created.
- When a switch is added to a group:
  - If the switch type is set to 'Toggle' change it to 2POS. Do not show the 'Toggle' option for the switch type.
  - If any other switch in the group is turned on, then turn off the newly added switch
  - If the group is always on, and no switch in the group is on, then turn on the newly added switch
- When a switch is removed from a group:
  - If the group is always on, and no other switch in the group is on, turn on the first switch in the group (if there are any)
- When a group is changed to always on:
  - If no switch in the group is turned on, turn on the first one.
- When the switch start position is set to UP or DOWN
  - If the switch is in a group then set the start position of all other switches in the group to 'Last Value'
- When a switch type is set to 'None'
  - Hide the group selector and always on checkbox

### Steps To Reproduce

Setup custom function switches with the various states described above.

### Version

Other (Please specify below)

### Transmitter

Jumper T20 V2

3djc, 2024-05-11:
Why can't you have a toggle in a group, if group is not always on ?

i would like to propose an alternative: have initial setup by group, instead of switches

GroupY: = (then all are remember), SWx (x is up, then rest in the group down), none (only if not always on)

philmoz, 2024-05-11:
> Why can't you have a toggle in a group, if group is not always on ?
> 
> i would like to propose an alternative: have initial position setup by group, instead of switches GroupY: = (then all are remember), SWx (x is up, then rest in the group down), none (only if not always on, all start down)

Good points, thank. I've updated the description above.

philmoz, 2024-05-11:
I would also suggest changing the B&W UI and moving the startup state setting to the end of each switch line and moving the always on checkboxes to a single line where the startup settings currently are.

3djc, 2024-05-12:
I really think the group way is much better and easier to understand for users

philmoz, 2024-05-12:
Something like this?
There is a section for each group where the always on and start switch is set.
When switches are in a group the startup position is hidden for the switch.

[image]

3djc, 2024-05-12:
YES YES YES !

PR #4994 (2024-06-01): fix: enforce custom switches rules when configuring
Fixes #4990 

Changes to the UI for setting up custom function switches to prevent invalid setup.
closed 2024-06-01 by pfeerick