EdgeTX/edgetx #4995: Taranis X9D Plus (2013) - startup screen (pre-splash screen!) scrambled / showing 'white noise'

RipperDrone, 2024-05-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Confirmed with 2 other users on Discord: Taranis X9D Plus (2013) screen is showing 'white noise' screen when powering up, regardless whether splash screen is switched on or off (splash screen will show normally if switched ON, right after the 1-2 secs 'white noise screen'. Looks like screen memory buffer holds random data by the time screen is switched on.

### Expected Behavior

up to v 2.9.3, startup behaviour was normal, no white noise screen shown on power up.

### Steps To Reproduce

1. Flash ETX2.10 to transmitter
2. Switch transmitter on

### Version

Other (Please specify below)

### Transmitter

Other (Please specify below)

### Operating System (OS)

Windows

### OS Version

11 x64

### Anything else?

[image]

pfeerick, 2024-05-12:
> Since it seems to be confined to Taranis transmitters and everything was
> running fine up to v 2.9.3, I tried to spot recent changes and found this
> refactoring commit by @raphaelcoeffic <https://github.com/raphaelcoeffic>
> to be related to display management - maybe it's related somehow.. 5153aae

It's not just white noise, it can also be last frame that was displayed if
you turn back on again after just turning the transmitter off (shutdown
image). It affects more than taranis also, I have seen that on T20 (i.e.
reverse OLED). As if the display buffer not cleared before/after enabling
display.

pfeerick, 2024-07-04:
There may have been two separate issues at fault here... X9D and X9D+ would have suffered from the screen not being cleared on startup, hence whatever was in the buffer was being shown, hence "garbage", and by happy coincidence that has been fixed while fixing a completely different issue for the Commando8. The T20, etc should never have been affected by that bug, but currently don't exhibit that behaviour, so will consider that resolved also unless it pops up again.

closed 2024-07-04 by pfeerick

RipperDrone, 2024-07-07:
confirmed, resolved on my T16S Plus (2013). Thx again!

THEthatdude, 2024-07-14:
I'm still getting white snow as well on my X9D+ running 2.10.2. Radio seems to work fine despite it. Custom splash image comes after the snow (about 2 seconds)

pfeerick, 2024-07-18:
This will be fixed  in 2.10.3 as a side-effect of #5245... X9D/X9D+ weren't doing the screen clear.

PR #5245 (2024-07-04): fix: Commando8 does not need the startup animation
Fixes #4798 since startup animation pwrOff() was responsible for shutting down radio
Fixes #4995 since screen was not being cleared on power up for radios with power switches