EdgeTX/edgetx #4998: Input Source - discrepancy between color and B&W radios

philmoz, 2024-05-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

@3djc discovered that color and B&W radios handle the Input Source field differently - they apply different filters to determine which sources are actually value for an Input.

Color radios use a function called isSourceAvailable(), B&W use isSourceAvailableInInputs().

This looks to date back to the initial Lvgl conversion around 2.4/2.5. In 2.3 both used the isSourceAvailableInInputs() function.

The table below shows the difference I can find between these two filters.

| Source | Color | B&W |
| --- | --- | --- |
| HELI | YES | NO |
| SPACEMOUSE | Only if configured on serial port | YES |
| Channels | Only if Mix exists for channel | YES |
| TIMERS | YES | NO |
| BATT, TIME, GPS | YES | NO |
| Telemetry | ALL | Only if UNIT < DATETIME |

Since isSourceAvailableInInputs() was originally used by all then the color change may be a mistake not picked up.
Or it could be a deliberate change - I don't know the history.

Asking for feedback on how to handle this:
- Do nothing, leave the status quo
- Change color to use isSourceAvailableInInputs()
- Change B&W to use isSourceAvailable()
- Some other solution

### Expected Behavior

Consistency between color and B&W radios.

### Steps To Reproduce

Compare options available for input source on color VS B&W radios.

### Version

2.9.4

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-05-13:
Blame history suggests it was changed in 2.5 as part of a bug fix. This is
pre LVGL...

https://github.com/EdgeTX/edgetx/pull/739

Although `isSourceAvailableInInputs` exists, in the first instance I
personally would prefer much of the current behaviour to stay, so would be
more in favor of ditching it entirely in favor of isSourceAvailable.

3djc, 2024-05-13:
Are some of those source even possible ?? What happens if you put a timer as a source ?? How is range mapped to -RESX to +RESX
The original idea of isSourceAvailableInInputs() is to ensure only source valid (in terme of range) can be set as source

closed 2025-10-19 by pfeerick