EdgeTX/edgetx #5000: EdgeTx 2.10 TARANIS X9E strange input source list and data corrupted

MicheleVilla, 2024-05-13:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Upgrading Taranis X9E from 2.9.4 to 2.10 data models became corrupted, input source list show partial and wrong list
same behavior in Companion simulation also starting with new model

### Expected Behavior

By updating the firmware the models data must not be corrupted and in source input list all input must be showed correctly
In radio and in companion also

### Steps To Reproduce

Upgrade firmware from 2.9.4 to 2.10
use companion simulator
go to input or mixer menu and choose an input source

### Version

Other (Please specify below)

### Transmitter

FrSky X9E / X9E Hall

### Operating System (OS)

Linux

### OS Version

Linux Mint 20

3djc, 2024-05-14:
Confirmed, affects all 212 radio in some way

MicheleVilla, 2024-05-14:
Please check also the custom screen display visualization
sometimes number cover the string description

3djc, 2024-05-14:
I found an issue, but I'm not too sure it is the one you mentioned, since your description lacks precision.

If there are other issues, please open separate issues, and please include screenshot or photo, so we can see exactly what you are referring too

3djc, 2024-05-14:
Could you check this fixes fixes the issue you found ?

[X9E-TEST-5000.zip](https://github.com/EdgeTX/edgetx/files/15305048/X9E-TEST-5000.zip)

MicheleVilla, 2024-05-14:
Tested
it work but apparently LR seems to be swapped with S2 and LS with S1

pfeerick, 2024-05-14:
TR# are trainer channels

PR #5002 (2024-05-15): fix(bw212): wrong source string display
Issue in #5000 lacks details, but I was able to find an issue in the area described

This fixes : #5000

For 2.10.1 too please :)
closed 2024-05-15 by pfeerick