EdgeTX/edgetx #5006: EdgeTx 2.10 Taranis X9E F3 F4 missing in Hardware configuration

MicheleVilla, 2024-05-14:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In hardware configuration menu F3 and F4 Pot Input have disappeared while they are physically present on the PCB

### Expected Behavior

F3 and F4 must be visualized in Hardware Menu

### Steps To Reproduce

Go to Hardware menu by long press key Menu
check if F3 anf F4 Pot are visualized

### Version

Other (Please specify below)

### Transmitter

FrSky X9E / X9E Hall

### Operating System (OS)

Linux

### Anything else?

[image]

pfeerick, 2024-05-14:
Also in Companion, so may simply be the hwdef that is missing entries.

[image]
[image]

MicheleVilla, 2024-05-15:
as I have already had the opportunity to point out when I choose an input by moving the slider LR displays S2 and LS displays S1
By installing 2.10 the programs already present show malfunctions regarding these definitions
This can be seen using the radio
Could it be a HARDWARE DEFINITION problem (or only my problem)?

MicheleVilla, 2024-05-15:
> And I told you to open another issue. Adding issues in comments is the best way to get those NOT sorted.
> 
> Anyhow, could you try this: [X9E-TEST-5006.zip](https://github.com/EdgeTX/edgetx/files/15317594/X9E-TEST-5006.zip)
I thought it was related to this issue, for the future I will open a new issue for every single problem as you suggest
I will test your solution this evening

Thank you

3djc, 2024-05-15:
[X9E-TEST-5006-2.zip](https://github.com/EdgeTX/edgetx/files/15317756/X9E-TEST-5006-2.zip)

PR #5012 (2024-05-19): fix(x9e): missing F3/F4 pots, slider hwdef changes
Match declaration with OTX initial source

Fixes #5006
closed 2024-05-19 by pfeerick