EdgeTX/edgetx #5010: Option for BLUE power LED

bob01, 2024-05-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

2.10 changed the power LED color on the TX16S from BLUE to GREEN - I understand the reason given but that said I really prefer (as do most guys I fly with) prefer the BLUE.

GREEN makes the radio feel old.

### Describe the solution you'd like

Revert to BLUE for power / transmitting or provide an option for power LED color.

pfeerick, 2024-05-15:
That is a bit of a problem, actually, as the TX16S is one of the radios that should *not* have changed colour... both it and the LR3PRO were supposed to be blue after startup (i.e. those two are reversed), so need to see why that changed. However, ideally there would be a option to set the warning, startup and operational colours, as everyone has different preferences, as well as helping if color blind, etc.

Model-Man, 2024-05-15:
The blue led was so much better and I asked the very same question on Discord. Now the power led has gone green it is also the same colour as the charging led which is a little confusing. If a simple colour change slider could be added in the TX settings so the user could pick and choose, it would be so much better.

MRC3742, 2024-05-15:
Also effects the Jumper T16 transmitter.
Upon startup, Power LED is Blue, after splash screen delay the LED shuts off completely.
v2.10.0 Release

Rich

3djc, 2024-05-15:
The current sequence is blue during boot up, red on errors, and green when everything is ready 

While the last color can be changed by setting, the initial green cannot as settings are not available at that time. So it an errorless start, there would be no color change when ready

Model-Man, 2024-05-15:
If this was the case why was blue the fully loaded up colour in 2.9.4 ?

3djc, 2024-05-15:
Because no real standardisation was taken into account it was just random. Then the manufacturers asked for all their radios to behave the same. Since the most universal color to say 'all is ok' is green, that was chosen

MRC3742, 2024-05-15:
According to the Jumper T16 schematic, I don't believe the  green LED is connected to the MCU.
So I get BLUE on Startup, RED on Error, and OFF after everything is ready.

Rich

Camo55, 2024-05-15:
Would the Power switch LED be able to be a separate RGB color selector similar to how the MT12 does it?

3djc, 2024-05-15:
No, and the green led turns on long before the radio is started enough to have rgb pulse ready

Model-Man, 2024-05-15:
So although this is effectively an open source software, the manufactures who I assume have several Edge TX programmers on the payroll, are now enforcing the format they want on the radios.

3djc, 2024-05-15:
First of all, I find it an amazing tribute to our work that the main issue you have with our software is the color of a led on the radio.

If manufacturers had any of us on their payroll, they would do exactly what frsky did once they managed to hire one of the otx devs: stop open source and do their own. Why would they pay anyone to have their work published as open source ????

Can you really argue that harmonising the color of power led across radio is a bad thing ? You seem to have a preference for blue, nothing wrong with that, but it's just that, a preference, while there is a rationale to have it green.

That said, the door is opened for a setting, I was merely stating that the first two led color cannot be changed based on user preference, simply because technically, the user settings cannot be read until sd is mounted and ready, and that happens very late in the process.

pfeerick, 2024-05-15:
> Also effects the Jumper T16 transmitter. Upon startup, Power LED is Blue, after splash screen delay the LED shuts off completely. 

Yeah, need to check what LEDs that radio has which are MCU controlled, green could be charge only. On the TX16S I think it is both MCU and charger controlled, but if it's charging we can't turn the green one off, which is one reason blue I think blue was the preferred operational colour for that radio.

> Then the manufacturers asked for all their radios to behave the same. Since the most universal color to say 'all is ok' is green, that was chosen.

No, that is not correct. We as a project chose to standardise the colours, as they were _somewhat_ standardised already, but new radios were being added with different colours, or GPIO definitions wrong because of colour swaps in the code (e.g. #2249, #2346, #3201). Then manufacturers were asked if they had a preference for pre-existing radios - i.e. RM for TX16S, BetaFPV for LR3Pro, in those cases they both requested it be left Blue once operational. Then there are some radios such as the X9D+ and Commando8 which only have blue or green, etc. Even a comment in the code that hints at this:

https://github.com/EdgeTX/edgetx/blob/c2e7d9153996a627ef254344cf5f0bf80cce9147/radio/src/opentx.h#L598-L611

However, it looks like https://github.com/EdgeTX/edgetx/pull/3715 undid part of this. So it looks like the todo from #2346 is probably needed now... "Future todo - have each radio raise a flag for blue over green status color". Was a chore I'd identified to work on for 2.9 (https://github.com/EdgeTX/edgetx/pull/2249#issuecomment-1236894642), as well as getting rid of the radio specific defines in that area and instead have it as part of the target options... but obviously didn't get there... 

> So although this is effectively an open source software, the manufactures who I assume have several Edge TX programmers on the payroll, are now enforcing the format they want on the radios?

Just because this is an open source project, doesn't not mean you _just_ get what _you_ want. But it does mean you can download/modify the code, compile it yourself. We'll even help you by telling you what code to adjust if you ask nicely, and you don't even need to download the compile environment since we went to the trouble of making it possible to build it using a cloud IDE. It is the Project Steering Committee (PSC) that determine what happens generally, and we are all private individuals who are donating our time and expertise to the project. I raised the concern that stuff was all over the place, and that it should be standardised for consistency and there was consensus on this point, with it coming mostly (if not completely) into effect in 2.8.

philmoz, 2024-05-15:
> On the TX16S I think it is both MCU and charger controlled, but if it's charging we can't turn the green one off, which is one reason blue I think blue was the preferred operational colour for that radio.

My understanding is that the TX16S was supposed to be green for operational; but due to a bug in the code it only changed the color if there was a throttle check on startup. This was fixed in 2.10.

Model-Man, 2024-05-15:
Many thanks for the detailed reply. I have only admiration for the work everyone does with Edge. I guess it must be hard to please everyone, as well as trying to standardize code across a multitude of different TX models.

pfeerick, 2024-05-16:
> My understanding is that the TX16S was supposed to be green for operational; but due to a bug in the code it only changed the color if there was a throttle check on startup. This was fixed in 2.10.

No, the RM TX16S was specifically requested to remain blue. Are you sure it was that, and that it wasn't that it was *stuck* green, and never changed to blue? Because the define line was changed in https://github.com/EdgeTX/edgetx/pull/3715 from being radio specific, to using the manufacturer defines, it undid the radio specific configuration.

philmoz, 2024-05-16:
On 2.9 if you start with a throttle check warning the LED is changed to green after you clear the warning - the call to set the led was only being done if there was an error. B&W radios alway set the led color regardless.
When I queried this the consensus was it was an error and should always set the color - the defined color for TX16S when error is cleared is green.

pfeerick, 2024-05-16:
I think you are talking about something different... yes, there was an issue with the led colour not actually being **set** on colorlcd when clearing a warning. However, this is nothing to do the underlying define which determines those colors, which was changed from #3715 ... which changes the TX16S (and other radios) from blue/blue to blue/green. 

[image]

mhk1058, 2024-05-16:
> Because no real standardisation was done initially, it was more or less was just random (just red was enforced) Then the manufacturers asked for all their radios to behave the same. Since the most universal color to say 'all is ok' is green, that was chosen

Ironically my quads' (using iNav)LEDs show green when disarmed (safe) and blue when armed. Another thumbs up from me to at least have an option to have blue.

GeneralPancakeMSTR, 2024-08-30:
Another vote for Option for Blue, or better yet, be able to turn off the LED entirely.

eXeCy, 2024-11-08:
I personally prefer the blue color too.. Green is getting me crazy and is really ugly.
If you please compile a FW with always blue it will be much appreciated.

mhk1058, 2024-11-08:
I think it's the manufacturers working group that you need to appeal to,
they set the standards for what colour means what across the hardware, not
the EdgeTX guys and gals. The blue colour was always a glitch in the matrix.

gagarinlg, 2024-11-08:
As far as I know there is no manufacturers working group.

3djc, 2024-11-08:
The manufacturers did ask to standardise color across their range, which only makes sense. Red was chosen for error as it is widely used this way. Green for 'ok' for the same reason, that left blue for initialisation in progress

gagarinlg, 2024-11-08:
When people complain about LED colors, everything else must be almost perfect 😉

etocii, 2024-12-20:
Option for color please!

JClaudeB7500, 2025-01-27:
I guys,
I also prefer that the LED stay blue during normal usage.
Thanks.

ElectricRoxy, 2025-02-04:
Green, Blue or any color is fine, but can we dim it down? We can reduce the backlight intensity, but not the power led, and doing crawling with the MT12 in the dark it is quite distracting

closed 2025-02-19 by pfeerick

relentlesstech73, 2025-07-24:
First time updating my TX16S Max in almost a year, and at first I thought my radio was broken after the update due to not having the blue power light like I am used to (I am green colorblind, so to me there is no light at all on to denote that the radio is even on) - My wife had to tell me that there WAS a light on, but it was green (Which I can't see) - Now my only options are to either revert back to an older version to get the blue light back, or try to find another radio that has changeable color support for colorblind individuals, which is quite disappointing

pfeerick, 2025-07-24:
Use the CloudBuild option of EdgeTX buddy to create a firmware with POWER_LED_BLUE set to N (for no).

https://buddy.edgetx.org/#/flash?source=cloudbuild&version=v2.11.2&selectedFlags=power_led_blue%3DNO&target=tx16s

relentlesstech73, 2025-07-25:
> Use the CloudBuild option of EdgeTX buddy to create a firmware with POWER_LED_BLUE set to N (for no).
> 
> https://buddy.edgetx.org/#/flash?source=cloudbuild&version=v2.11.2&selectedFlags=power_led_blue%3DNO&target=tx16s

Unfortunately, this didn't do anything, my wife says the LED is still green after re-flashing the CloudBuild

etocii, 2025-07-25:
> Unfortunately, this didn't do anything, my wife says the LED is still green after re-flashing the CloudBuild

I found that set power_led_blue to yes works for me :)

relentlesstech73, 2025-07-25:
> I found that set power_led_blue to yes works for me :)

Perfection! I can see the light now! Thank you, good sir!

Unoqueva, 2026-09-16:
The Jumper T16 only has red and blue LEDs. With the default firmware, the one you download via Companion or from GitHub, no LED lights up to indicate the radio is on when it is powered up, simply because there is no green LED.

I wonder if a line of code could be added so the compiler automatically generates a build for the T16 with the "Power_led_blue" option set to YES.

My power LED hadn't been working for a while, and it was driving me crazy until I found this; I assume I’m not the only one who has experienced this without knowing the reason.

J-Sorenson, 2026-09-16:
> The Jumper T16 only has red and blue LEDs. With the default firmware, the one you download via Companion or from GitHub, no LED lights up to indicate the radio is on when it is powered up, simply [...]

Please read the comments above: they added a flag to EdgeTX Buddy to create a build with it set to Blue.  Now, I don't know if anyone with a T16 has tried it, but give it a go. This is for Jumper T16 EdgeTX v2.11.7:
https://buddy.edgetx.org/#/flash?source=cloudbuild&version=v2.11.7&selectedFlags=power_led_blue%3DYES&target=t16

pommecrafter, 2026-10-31:
I've made a custom software tutorial to change the Power button color (no physical mods) [Here](https://www.reddit.com/r/fpv/s/ARtHm7jN2t) (reddit link)

pfeerick, 2026-10-31:
Just a FYI, 3.0 will allow you to define the error, transmit and ready colours at runtime, so there will no longer be a need to set it at compile time / do a special build

OzDruiD, 2026-10-31:
> 31\. Just a FYI, 3.0 will allow you to define the error, transmit and ready colours at runtime, so there will no longer be a need to set it at compile time / do a special build

This is amazing, thank you!   Is there a PR to follow and\or track when its added to master to test?

pfeerick, 2026-10-31:
Been in the nightlies for about the last month... #7806

OzDruiD, 2026-11-08:
Finally got to try this  Will take a backup and try my production\daily driver my MK3  - just realised it wont work on my "Test" backup radio as its not a H7.

pfeerick, 2026-11-09:
> just realised it wont work on my "Test" backup radio as its not a H7.

Not completely, no...  the brightness part won't work if it's not H7, but the basic colour option will work for almost all radios ;)

OzDruiD, 2026-11-09:
> > just realised it wont work on my "Test" backup radio as its not a H7.
> 
> Not completely, no... the brightness part won't work if it's not H7, but the basic colour option will work for almost all radios ;)

Interesting, ok will try and reflash my TX16S MK2 with nighly again,  the Color LED section under Ratio Settings wasnt even showing up.

PR #5922 (2025-02-19): feat: allow for compile time user override of power LED color
Resolves #5010

Summary of changes:
- allows for user override of Power LED startup sequence from the mostly standardised RED (pre RTC init), BLUE (starting up), GREEN (ready to use) via compile time flag
- inverts the blue -> green sequence to be green -> blue, allowing for blue to be the "ready" color, and adds mention of this override to build opts listing. 

Tested on TX16S, Boxer and PL18EV