EdgeTX/edgetx #5014: BetaFPV Literadio 3 pro time not saving

edvard5, 2024-05-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Whenever I set time in radio settings it resets when power cycle happens

### Expected Behavior

For time to not reset

### Steps To Reproduce

1. Set correct time
2. Turn off radio
3. Turn on radio
4. The time is back to 2000-01-01 00:00:00

### Version

2.10.0

### Transmitter

BetaFPV LiteRadio3 Pro

### Operating System (OS)

Windows

### Anything else?

Same thing on both versions 2.9.4 and 2.10.0
RTC battery checked with multimeter, shows 3V.

pfeerick, 2024-05-16:
I just flashed both the firmware and bootloader for 2.10 to my LR3PRO (had been running a mix of a 2.10RC bootloader, and a 2.11 nightly) and it initially lost the time, but saved it when I set it. I have cycled it multiple times, and time is fine, and is saved if I change it again. RTC battery on mine is reported as being 2.98V

edvard5, 2024-05-19:
> I just flashed both the firmware and bootloader for 2.10 to my LR3PRO (had been running a mix of a 2.10RC bootloader, and a 2.11 nightly) and it initially lost the time, but saved it when I set it. [...]

For me if I set it it stays untill I turn the radio off. When I turn it on its 2000-01-01 really frustrating :( Hope it's not hardware related issue