EdgeTX/edgetx #5015: SD card recognition issues

OlivierBoubou, 2024-05-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After updating Edgetx to version 2.10.0, I have the message NO SD CARD with small capacity SD cards (512Mb, 1Gb)

### Expected Behavior

Small capacity SD cards are recognized

### Steps To Reproduce

After update Edgetx from version 2.9.4 to 2.10.0

### Version

2.10.0

### Transmitter

FrSky Q X7 / Q X7S (ACCST)

pfeerick, 2024-05-16:
This may not be related just to small cards, but instead cards that are extremely slow to respond to the initialisation request on startup and have exceeded the timeout. Have you tried with a name brand SD card such as a Sandisk or Samsung one as a comparison?

3djc, 2024-05-16:
The low levels drivers for SD have been updated, it was absolutely required as part of a much bigger change (to drop StdPeriph), which results in some card that used to work not working anymore, but also some card that didn't now do work.

Unfortunately, due to the radio hardware, those SD card are read on  BW radio in a way that is least used in the market, so  many manufacturers don't test their card in this mode.

hylaride, 2024-05-18:
I'm also having an issue on my Boxer with 2.10 and a 32GB Lexar SD card Both the boot loader and 2.10 itself don't see the SD card.

nanodude, 2024-05-19:
I'm also having the same problem with a Toshiba SD card on a Taranis X9D which worked fine prior to 2.10.0.

kasatka60, 2024-05-20:
Jumper T12 Pro. 2.10 - no cd card. 2.9.4 - all good

msun12000, 2024-07-27:
Possible solution - I had a number of SD cards which were not recognised by my TX (Frsky X9 Lite S). The solution for me was to format the SD card using Rufus and choosing "Non bootable" under "Boot selection". I assume that the problem was due to come MBR or partition table corruption or variance which does not affect Windows but does affect TXs. I hope that helps.

kasatka60, 2024-08-22:
> Possible solution - I had a number of SD cards which were not recognised by my TX (Frsky X9 Lite S). The solution for me was to format the SD card using Rufus and choosing "Non bootable" under "Boot selection". I assume that the problem was due to some MBR or partition table corruption or variance which does not affect Windows but does affect TXs. I hope that helps.

It didn't help me.

nightflasher, 2025-03-07:
Wanted to change my sdcards (old 2GB sandisk) in my three tx (jumper t-lite v1 4in1, v2 ELRS, NV14) and bought 3x intenso 4GB class-10 and only one worked, but only empty and no contents at all. After copying all the contents to the new cards, all radios say "NO SD CARD".

Funny thing: jumper t-lites sdcards are formatted without a partition (like sdc instead of sdc1) in vfat with name/label and they are very fast compared to partitioned sdcards.

r1000ru, 2025-06-26:
I can't upgrade my Jumper T-Pro from version 2.9. The solution turned out to be to use card SanDisk Extreme Pro V30 U2 A and Rufus to format the card in FAT32(extended, not ExtFAT!). EdgeTX 2.11.2 has started

3djc, 2025-11-14:
The most likely to work are Class 10, non endurance, original from main brands (Samsung, Lexar,...). But there are tons of fakes, even from shops as popular as Amazon or the likes

pfeerick, 2025-11-15:
Supply chain injection/fakes is a real thing, but it is less likely if you get from Amazon  and ensure you are getting it from the official store for the manufacturer, rather than a random seller. Or better still, a local retails where you can take it back to wnd lodge a complaint ;) 

Regardless of where you by from, using a tool like https://www.grc.com/validrive.htm (windows)  or https://sascha-simon.com/en/capacity.html (macos) or f3 (Linux) with a new USB or SD card won't go astray (unless you're comfortable with finding out at the worst possible time it's a fake)... ;)