EdgeTX/edgetx #5028: add trim to subtrim no longer zeros trims

driedeker, 2024-05-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I mabe wrong but I thought add trim to subtim used to zero/centre the trims but not seeing it on my X9D+2019

### Expected Behavior

trims to recentre after adding to subtim

### Steps To Reproduce

use the option to add trim to subtrim

### Version

2.10.0

### Transmitter

FrSky X9D+2019

### Operating System (OS)

Windows

### OS Version

windows 11

HippoDan, 2024-05-29:
Still works correctly on 2.10 TX16s Mk2.  This may be radio model specific.

pfeerick, 2024-05-29:
I can't reproduce this on X9D+2019... pressing and hold on `Trims => Subtrims` to apply the setting correctly adds the trims to the subtrims, and resets the trims to neutral. Have you been able to reproduce this again @driedeker

driedeker, 2024-05-31:
still not working for me. perhaps I am not doing it right what are you doing to copy the trims to subtrims .which way do you do it. I am going to outputs holding enter up pops the option to copy to subtrim hit enter go back to main munu and trims have not changed. now on 2.10.1

driedeker, 2024-05-31:
Ok now I have found the copy to trim heading in outputs and that worked. what I was doing was as soon as I got to outputs hold enter up poped a menu with copy trim to subtrim enter on that and that did not work.

pfeerick, 2024-05-31:
Ah, ok, that option is different. It only copies/adds (duh!) the trim to
subtrim, not moves, hence is working as intended.

closed 2024-05-31 by pfeerick