EdgeTX/edgetx #5033: Ransomware update

orstratcat, 2024-05-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

After opening companion when downloading the current files is when the Ransomware is detected. Then Malwarebytes deletes companion.exe file and logs the file

### Expected Behavior

no malware alert

### Steps To Reproduce

open companion and select download components
[This intiates malware block.odt](https://github.com/EdgeTX/edgetx/files/15366091/This.intiates.malware.block.odt)

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

win11

### Anything else?

[This intiates malware block.odt](https://github.com/EdgeTX/edgetx/files/15366092/This.intiates.malware.block.odt)

orstratcat, 2024-05-18:
I scanned both the Companion 2.10.0 download and the extracted files with Avast and Malwarebytes and found nothing. But when using the Companion update components button upon downloading the C480x272 it asked I wanted to download it again. I pressed no and Malwarebytes alarmed and blocked Malware.Ransom.Agent.Generic and deleted the Companion.exe file.

3djc, 2024-05-18:
Most likely a false positive

orstratcat, 2024-05-18:
Didn't get this Malware Alarm with 2.9.4

3djc, 2024-05-18:
Especially funny since this zip file doesn't include anything that can be run on a PC :D

orstratcat, 2024-05-18:
Okay I uninstalled Malwarebytes and reinstalled it, no more Ransomware issue! So it seemed to be a Malwarebytes glitch! Problem Solved!

closed 2024-05-18 by 3djc

pfeerick, 2024-05-19:
In addition to being built in a clean build environment, the files are scanned before release on virustotal... so when there is only only 1 (or 3) out of 70 virus/malware scanners complaining, can be pretty sure it is a false positive. I have Malwarebytes go stupid in the past also. 

win32 zip - https://www.virustotal.com/gui/file/571227b9de412897d705fd6aa17eb9b252391cbd373dedc96a40ded815636833
win32 installer- https://www.virustotal.com/gui/file/33637d9c7423b9e78cb36c72ad0ecb10ae1f676eb05475ffd3ba657e98d70e45

win64 zip - https://www.virustotal.com/gui/file/f517fa6821a4adc7355e87a5828194ab2b12bea3e62510ed95d5a53c88ef7f03
win64 installer - https://www.virustotal.com/gui/file/1b87e58dcc6f6bdca884f4b5e538b562a69c53f650c3d9bce129c33e0281b1ea