EdgeTX/edgetx #5035: The screenshot of the TX16S is shifted sideways

koaxheli, 2024-05-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

With the transmitter, the screenshot in xxx.bmp format is shifted sideways. 

### Expected Behavior

With Companion in xxx.png format it is OK.

### Steps To Reproduce

1. make a hard copy on the transmitter
2. Open the image in the PC and you will see the offset
3. make a copy in Companion and it is OK
4. the difference is that with the TX16S the BMP is not saved correctly.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### Anything else?

[screen-2024-05-19-145003.zip](https://github.com/EdgeTX/edgetx/files/15369589/screen-2024-05-19-145003.zip)
[screenshot_tx16s_24-05-19_15-01-58.zip](https://github.com/EdgeTX/edgetx/files/15369590/screenshot_tx16s_24-05-19_15-01-58.zip)
[screen-2024-05-19-145003.zip](https://github.com/EdgeTX/edgetx/files/15369591/screen-2024-05-19-145003.zip)
[screenshot_tx16s_24-05-19_15-01-58.zip](https://github.com/EdgeTX/edgetx/files/15369592/screenshot_tx16s_24-05-19_15-01-58.zip)

3djc, 2024-05-19:
Not sure what your issue is. I compared the two pictures using https://www.diffchecker.com/image-compare/ and the only difference seems to be SJ position

[image]

Salto42, 2024-05-20:
I can confirm that the screen image is not unpacked.
If you go to the "SCREENSHOTS" folder and select a screen photo, you'll see the offset in the preview.
(TX16s , 2.11.0)

Salto42, 2024-05-20:
I realized this when I tried to use “Vu”(https://www.schleth.com/fpv/vu-a-simple-image-viewer-for-edgetx-radios-with-big-screens-2113.html)