EdgeTX/edgetx #5036: In the mixer menu, only the - sign is displayed at >-100%.

koaxheli, 2024-05-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

If the weighting is to display a value of -100%, there is not enough space so that only the - sign ap

### Expected Behavior

Please make the field larger

### Steps To Reproduce

1. go to the mixer settings
2. change the weighting to -100%
3. only the minus sign is displayed

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Anything else?

[SCREENSHOTS.ZIP](https://github.com/EdgeTX/edgetx/files/15370140/SCREENSHOTS.ZIP)

3djc, 2024-05-19:
Seems to be ok here

[image]

philmoz, 2024-05-19:
I can reproduce this in PR 5031 (fixed now); but not in 2.10 and 2.11 main branch.

@koaxheli If you are using a nightly build or the build from a PR please indicate this in the issue description.

pfeerick, 2024-05-20:
Screenshot attached looks more like from a nightly than 2.10  - notice the broader selection outline.

koaxheli, 2024-05-20:
The screenshot is from version 2.11.0
[image]
[image]

pfeerick, 2024-05-22:
Thanks for confirming. You selected 2.10 rather than nightly/other hence the confusion there ;)

philmoz, 2024-06-12:
This is now fixed in 2.11.

closed 2024-06-12 by pfeerick