EdgeTX/edgetx #5043: Companion 2.10 Simulator for RM Pocket - Switches incorrect

MirlynG, 2024-05-21:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

In Companion Simulator for RM Pocket, switches SA and SD are momentary when they should be toggles. Albeit they can be locked. Switch SE is a toggle when it should be momentary.

### Expected Behavior

Expect switches SA and SD to be toggles and SE to be momentary

### Steps To Reproduce

Open relevant model in simulator and operate SA,SD and SE. Problem will be seen

### Version

2.10.0

### Transmitter

RadioMaster Pocket

pfeerick, 2024-05-21:
I can't reproduce this with Companion 2.10 on Windows. When starting from a blank config, as well as reading back from the radio, A and D are 2POS, B and C are 3POS, and E is a toggle (2POS toggle) - which is the correct stock configuration. 

i.e. this is the correct configuration for the radio. 

[image]

MirlynG, 2024-05-21:
Sorry. I used wrong naming convention for the switches. Your hardware screenshot is how my tx physically behaves, but you will see from my attached picture that my hardware screen is not the same. This is shown on the tx and in simulator with a brand new base model!

I have also uploaded a video here showing the actions here:
 https://drive.google.com/file/d/1xLrrPaGhbY79b0nzvvIGs7steTif4eQY/view?usp=sharing
Obviously watch SA then SD and then SE

[image]

MirlynG, 2024-05-22:
I forgot that you can change the configuration of the switches in Hardware. Now done to reflect what they should be and will run it in Companion tomorrow. I think I may have wasted your time a bit, sorry, but am grateful for your input which pointed me in the right direction. 
You guys are amazing for what you do for us mere mortals. Thankyou so much.

MirlynG, 2024-05-22:
OK. That has sorted this issue but I have two more which I will post shortly
Again, many thanks

MirlynG, 2024-05-22:
OK. That has sorted this issue but I have two more which I will post shortly.
Again many thanks