EdgeTX/edgetx #5044: Longer filenames....

HippoDan, 2024-05-21:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Even on older OpenTX builds for my old Taranis, I can use long filenames for sounds.  Having just gotten a TX16s Mark 2, I am shocked that it only supports 6 character names!  Touch screen, but no filenames?  I copied my older models over an find that many of my custom sounds will have to be renamed, and the models edited.  Going forward, is there a real reason for the short names o the TX16s?

### Describe the solution you'd like

Longer filenames up to the Fat32 limit should be supports, or at least anything longer.

HippoDan, 2024-05-21:
I just realized there's is no such limit for images.  So it is possible I guess.

philmoz, 2024-05-21:
I'm guessing you have a wide screen B&W radio like the X9D or X9E - these limit the sound file names to 8 characters, all others are 6 characters.

There is no reason the limit could not be increased to 8 for color radios. I will look at this later.

HippoDan, 2024-05-29:
> I'm guessing you have a wide screen B&W radio like the X9D or X9E - these limit the sound file names to 8 characters, all others are 6 characters.
> 
> There is no reason the limit could not be increased to 8 for color radios. I will look at this later.

X9D+ seems to allow 10 character sound names in OpenTX 2.3.11.

philmoz, 2024-05-29:
> X9D+ seems to allow 10 character sound names in OpenTX 2.3.11.

From OpenTx source code:
```C
  #define LEN_FUNCTION_NAME            8
```

Tested 2.3.15 simulator and only 8 characters allowed.

pfeerick, 2024-05-29:
As you can see, the files with 9 and 10 characters to the filename are not shown. 
[image]

2.3.11 should be no different
https://github.com/opentx/opentx/blob/b9f89eabf35185881da685ae05081719cba64ece/radio/src/dataconstants.h#L123

HippoDan, 2024-05-29:
Sorry,  that was my mistake.   I have some 10 char files on my SD, never realized they weren't showing up in the list of hundreds.

PR #5046 (2024-06-06): feat(radio): increase length of sound/script file name to 8 for SF/GF
Fixes #5044 

Wide screen B&W radios have a limit of 8 characters for the file name parameter (sound files and script files).
Color and 128x64 B&W radios limited this to 6 characters.

Color radios have plenty of memory so can easily handle an 8 character limit.

128x64 B&W radios allocated space for 8 characters even though only 6 where allowed - so increasing the limit does not use any more RAM.

A small tweak to the 128x64 UI allows the longest 8 character name to still fit on the SF/GF setup page.

Possible candidate for [... cut]
closed 2024-06-06 by pfeerick