EdgeTX/edgetx #5052: Timerzeit wird in unterschiedlicher Lautstärke wiedergegeben

WilliPe, 2024-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Ich frage TMR1-Value mit einem Taster ab. Timer-Value ist z.B. bei der Abfrage 7:30 Minuten. Bei der Sender-Ansage wird die Zahl "7" sehr leise, "Minuten" in normaler Lautstärke, "30" sehr leise, "Sekunden" in normaler Lautstärke angesagt. In den System-Einstellungen habe ich beide Wave-Lautstärken auf maximale Lautstärke gestellt.

### Expected Behavior

Mein Wunsch ist, die Ansage in gleicher Lautstärke wieder gegeben wird.

### Steps To Reproduce

Taste für TMR1-Value drücken: Fehler in der akustischen Wiedergabe des Wertes.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-05-22:
English?

WilliPe, 2024-05-22:
New issue i english

closed 2024-05-22 by WilliPe