EdgeTX/edgetx #5053: Companion 2.10 Simulator on Windows 10 for RM Pocket - shows wrong tx battery voltage

MirlynG, 2024-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

When opening a model on RM Pocket in Simulator, it shows the wrong battery voltage on the model page. i.e, radio voltage is 8.0 but shows a 6.7 in sim.
It also plays the low battery warning, even though the battery range is set to 6.6-8.4v.

### Expected Behavior

I'd expect that the simulator will reflect the radio voltage correctly.

### Steps To Reproduce

Open a model in simulator and see resulting error.

### Version

2.10.0

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### OS Version

10

### Anything else?

I have uploade a video here: https://drive.google.com/file/d/1DZsGEfZc3gTu2-D8ZLq2N17LX7roOa2u/view?usp=sharing

[image]
[image]
[image]
[image]

3djc, 2024-05-22:
How exactly do you expect the simulator to know your radio battery voltage?

MirlynG, 2024-05-22:
This was all done immediately after reading from the tx and saving the .etx file. I would expect the voltage on Companion to be the same as it was at that time. I realise it won't change.

As for the alarm, my mistake.

3djc, 2024-05-22:
Clearly not how it is designed to work, so not a bug. what you want is an enhancement then, but I'm not overly sure it makes a lot of sense. Maybe allow to change it on the telemetry simulation panel

pfeerick, 2024-05-23:
This is the same for all radios btw, not specific to the RM Pocket or Windows. It is a hard-coded value that with stock settings shouldn't trigger the battery alarm. Not sure if it _belongs_ in the telem simulation panel though, as it is a handset value, not telemetry... although there isn't really anywhere else to put it... i.e. it could be a simulator preference, but then that makes it hard to change if you want to change it dynamically... what about a dialog box off the tools menu?

Chris-1066, 2024-05-25:
I have a similar issue with the Companion 2.10 simulator (Mac OS) and a RM Boxer. 
On Companion 2.9 simulator the Boxer battery is set to 7.9V and doesn't trigger my LV alarm (7.2V for 2S lipo):

[image]

However, on Companion 2.10 simulator the Boxer battery is set to 6.5V and, hence, does trigger my LV alarm, which is annoying whilst trying to check out settings using the simulator:

[image]

To get around this I'm reducing the LV alarm setting to 6.4V in the radio settings and then trying to remember not to write these settings to the radio (I think 3.2V/cell is too low for a lipo).

As I understand it, the 7.9V/6.5V values are hard coded in to the simulator and cannot be changed by the user? Is there any other way to disable the LV alarm in the simulator without altering the radio LV settings that are sent to the radio?

beau76, 2024-07-23:
I am seeing the same problem with a boxer with all the default settings and the battery alarm goes off in the simulator

VancouverUmbrella, 2025-01-03:
I've just updated to Companion 2.10.5 from 2.9.4 and encountered this problem.

In 2.10.5, the simulated Tx voltage is 6.8V, which triggers my low voltage alarm of 7.2V.

In 2.9.4, the simulated voltage is 7.7V, which does not trigger the alarm.

I use the relatively high alarm of 7.2V because of the high power requirements of some ELRS Tx modules. In some cases, Dynamic Power can really such the juice from a transmitter pack.

VancouverUmbrella, 2025-01-03:
> This is the same for all radios btw, not specific to the RM Pocket or Windows. It is a hard-coded value that with stock settings shouldn't trigger the battery alarm. Not sure if it _belongs_ in the [...]

A simulator preference would work for me. In fact, that's where I looked to see if such a preference already existed. Personally, I have no need for the value to be dynamic.

PR #5763 (2025-01-12): feat(cpn): simulator allow manually setting transmitter battery voltage
Fixes #5053

Summary of changes:
- add Battery Voltage option to Tools menu

Notes:
- the radio will not update to the new voltage instantly but over a number of adc read cycles as filtered adc averaging is used.
- there maybe instances of rounding variance as the radio uses integers
- the tx battery settings for min max and warn values are only read on sim load ie changes via radio ui will not be detected
closed 2025-01-12 by pfeerick