EdgeTX/edgetx #5054: Simulator shuts down after a few seconds when selecting system button - Companion 2.10 and RM Pocket

MirlynG, 2024-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

After opening a model in Simulator and selecting the System button, the Tools window opens as expected. However, after a few seconds, Companion shuts down completely.
If you are quick and hit the "page forward" button as soon as Tools opens, you can then cycle through the rest of the pages without any problem. However, when you cycle through to Tools again and wait a few seconds, one of two things may happen. Either Tools will remain open as normal or, again, Companion may shut down completely. There does not seem to be any consistency in this.

### Expected Behavior

For Companion not to shut down after selecting System button.

### Steps To Reproduce

Open companion
Open an etx file
Select a model
Open Simulator
Click System button 
Wait a few seconds and Companion will shut down completely

### Version

2.10.0

### Transmitter

RadioMaster Pocket

### Operating System (OS)

Windows

### OS Version

10

### Anything else?

I have uploaded a video here:
https://drive.google.com/file/d/1DbUB1iCy0NcrFPk1x5Huo-HUlWgQLtag/view?usp=sharing

pfeerick, 2024-05-23:
I'm not on Windows atm, but I can't reproduce that with 2.10.0 Companion and RM Pocket simulator... 

Try enabling logging (fourth icon down on the side toolbar of the simulator) before going into the SYS/Tools screen, and see if anything promising shows up before it crashes out. If you don't see anything, enable debug logging in Settings -> Settings... -> Application Settings (tab). You'll see the path they are saved to (or can change it if you prefer) as soon as you enable either of those options. 

[image]

MirlynG, 2024-05-23:
Here you go Peter. Means nothing to me!!

 

image.png (view on web) <https://github.com/EdgeTX/edgetx/assets/5500713/8ea6e075-d645-4a20-b64b-c84e968aa2c7>

pfeerick, 2024-05-23:
Github does not accept email attachments. You'll need to go to this issue page and attach it there. 
[image]

philmoz, 2024-05-24:
Does it still crash if you move the SD folder to somewhere other than OneDrive/Documents?

[image]