EdgeTX/edgetx #5055: Timer time is played at different volumes

WilliPe, 2024-05-22:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I query TMR1 value with a button. For example, the timer value for the query is 7:30 minutes. When announcing the Tx, the number "7" is announced very quietly, "minutes" at a normal volume, "30" very quietly, and "seconds" at a normal volume. In the system settings I set both wave volumes to maximum volume.

### Expected Behavior

My wish is that the announcement is played back at the same volume.

### Steps To Reproduce

Press key for TMR1 value: Error in the acoustic reproduction of the value.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-05-22:
What sounds are you using ?

WilliPe, 2024-05-22:
de-Sounds (from Sound:System) (no self-created sounds)

3djc, 2024-05-22:
Post your sound folder, and your RADIO and MODEL settings

WilliPe, 2024-05-22:
Do I have to open the individual folders on the SD card and, for example,
copy the individual models? Or the individual sounds?

pfeerick, 2024-05-22:
You should be able to zip up the folders, and then drag that to the
comments section on GitHub. Don't attach to an email as attachments as that
doesn't work with GitHub.

WilliPe, 2024-05-22:
o.k., thanks. I do my best ...
Please wait.

WilliPe, 2024-05-23:
I zipped the sound and model folders. What data do you want from the radio
setup? I only see .bin and .yml files, which cannot be zipped.

3djc, 2024-05-23:
You need to post the zip file on github (not email). Please post one ZIP with your sound folder, and another one with RADIO and MODELS folders

WilliPe, 2024-05-29:
I found the solution myself. Please delete my issue. The reason was the
following: I didn't install the new sounds-file for 2.10 because I thought
I didn't need them. After I saved this sound file with my existing sounds,
the announcement now works completely at the same volume. Sorry that I
started an issue with this and thanks for the help.

closed 2024-05-29 by 3djc