EdgeTX/edgetx #5064: TX16S, EdgeTX 2.10.0, no S.Port telemetry from external module?

keithluneau, 2024-05-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

No sensors discovered on the model settings telemetry tab when using S.Port telemetry in the external module bay after upating to 2.10.0

It was working in 2.9.2, and also works after rolling back to 2.9.4

For context, I'm using a Dragonlink module connected to the external module bay using SBUS protocol, which feeds S.Port telemetry back to the S.Port pin in the module bay at 57600 baud.

### Expected Behavior

Sensors should be discovered like in previous firmware.

### Steps To Reproduce

Feed telemetry into the external module bay's S.Port pin.
Go to model settings, telemetry tab, delete any previously discovered sensors if present, try to discover new sensors and none are found.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

klaashartmann, 2024-06-04:
I had this same issue. Rolling back to 2.9.0 or 2.9.4 did not work. But the discussion here:
https://github.com/EdgeTX/edgetx/issues/3877
had me suspicious that the problematic change was introduced in 2.9.0. 

After rolling back to 2.8.0 it has worked seamlessly for me. I am not sure how it was working for the original poster in 2.9.0.

Blooskyy, 2024-06-19:
I have experienced a similar issue where certain telemetry sensors (eg altitude) aren't discoverable after updating to 2.10. However after restoring back to 2.9.4 they work perfectly. Not sure if this is an EdgeTX issue but without this working it won't be possible to upgrade.
TX16S

Amigache, 2024-08-01:
I have the same problem, after updating to 2.10.0, I cannot obtain telemetry information through the s.port of the external module, I am using Mav2PT and DL

keithluneau, 2024-08-01:
> I have the same problem, after updating to 2.10.0, I cannot obtain telemetry information through the s.port of the external module, I am using Mav2PT and DL

Unfortunately this still seems to be a problem, and with no response from anyone at EdgeTX I've resorted to rolling firmware back to 2.9.4 and sticking with that until it gets addressed.

nspector8, 2024-08-05:
I tried seeing if 2.10.3 may have corrected this issue, but still no telemetry in Yaapu.  

This is with TX16S and RFD TXMOD in external bay.

packetkiller, 2024-09-04:
I can also confirm that with RFD 900x v2 and TXMOD v2 radios both SBUS Telemetry and RC link are affected (not working) with EdgeTX 2.10.2. Works on EdgeTX 2.9.4.  Verified with 4 different sets of these radios. Seems to be an EdgeTX thing. I do get telemetry using TBS Crossfire radios using EdgeTX 2.10.2.

nspector8, 2024-09-24:
Attempted 2.10.4 with all fresh SD content and yaapu script, but still no fix.

RC link with SBUS and telemetry passthrough to MP is perfect, but zero telemetry in the radio.

TX16S Mk II
Matek H743-WLITE
RFD TXMOD v2.0
RFD 900ux v.2.0

Blooskyy, 2024-09-24:
I had solved the telemetry issue by increasing the baud rate from 115k to
400k on the TX16S.  Worth giving it a try.

https://www.expresslrs.org/quick-start/transmitters/tx-prep/#serial-baud-rate

nspector8, 2024-10-01:
Thanks for the suggestion Blooskyy!  Unfortunately though, adjusting my RFD900ux/TXMOD settings to 230400 and 460800, as well as the associated serial baud rate in the FC to the same, had no effect.  Also, since I'm using SBUS for RC link, there is only refresh rate available as a value to adjust in EdgeTX (default 7.0ms), and no adjustment for baud rate, unless there is a way to correlate that..

pierrotm777, 2024-10-01:
Try 400k and not 460,8k.

cardboardboks, 2024-10-02:
> Attempted 2.10.4 with all fresh SD content and yaapu script, but still no fix.
> 
> RC link with SBUS and telemetry passthrough to MP is perfect, but zero telemetry in the radio.
> 
> TX16S Mk II Matek H743-WLITE RFD TXMOD v2.0 RFD 900ux v.2.0

Same setup here with the same issue - TX16S Mk II Cube orange on carrier board RFD TXMOD v2.0 RFD900x
RC link with SBUS and telemetry passthrough to MP is perfect, but zero telemetry in the radio.
Rolled back to 2.9.4 as well and it works.

cardboardboks, 2024-10-02:
default was 400k I believe and I tried 115k as well, both no joy

robertlong13, 2024-11-22:
There seems to be some muddying the waters in this thread. @keithluneau 's issue is specifically that 2.10 cannot receive S.Port telemetry on the S.Port pin when using an external module using takes SBus for the channels.

This is the same issue reported in #3877, and fixed for 2.9.4 with #3986, but that fix did not go into main/2.10. In the discussion of #3986, @raphaelcoeffic mentioned that #3780 was a better fix in the long run, but notably, SBus is not touched in #3780.

I think (but did not think to test before I rolled back), that PPM with S.Port telemetry on the S.Port pin does work in 2.10, but those of us trying to use SBus with S.Port telemetry on RFD TXMods are out of luck.

@raphaelcoeffic, do I have this analysis correct?

nspector8, 2024-11-22:
@robertlong13,

To my knowledge, @keithluneau  is not using RFD hardware, but rather an ELRS external module.

i.e., this issue is not isolated to RFD users.

robertlong13, 2024-11-22:
@nspector8 
> To my knowledge, @keithluneau is not using RFD hardware, but rather an ELRS external module.
> 
 Nope, he says right in the OP that he's using DragonLink with SBus+S.Port, which as far as EdgeTX is concerned is the same as those of us running an RFD TXMod with SBus (like a few other people in this thread).
 
 There might also be some issue with external ELRS modules, but I don't think it's related to this issue.

keithluneau, 2024-11-22:
I'm actually seeing the problem when using a Dragonlink module. ELRS communicates with the handset via CRSF and isn't related.

When using Dragonlink, the handset sends data out to the module via sbus. The module has the ability to send telemetry to the handset using S.Port.

raphaelcoeffic, 2024-12-01:
@robertlong13 thx a lot for summarising the situation!

It would be nice if everyone could test #5689 and report if that works for you guys. You should now be able to select a telemetry protocol with PPM and SBUS external modules.

robertlong13, 2024-12-06:
@raphaelcoeffic, sorry for the delay. The fix worked perfectly on my X9D+2019

closed 2024-12-12 by pfeerick

klaashartmann, 2025-01-19:
I just tried this with the latest nightly build. 

I am using a RFDesign TXMOD v2.0. If I select SBUS then S.PORT appears in the dropdown box but I can't select it (it reverts to None). Telemetry still seems to appear but crashes Yaapu with "Error in refresh bad argument bxor ..."

If I select PPM + S.PORT the telemetry works perfectly (but of course there is no RC signal).

I have reverted to firmware 2.8.0 where it is working robustly.

xfacta, 2025-04-20:
#5689 still not merged as far as I can tell, and telemetry still not available when using an external module with SBUS output, such as the RFD TXMOD

pfeerick, 2025-04-20:
> https://github.com/EdgeTX/edgetx/pull/5689 still not merged as far as I can tell,

Er...  still not merged?

[image]

#5944 is the open issue to comment in now

xfacta, 2025-04-22:
Sorry it appears I commented in the wrong thread.
I saw the merged status put could not see this issue or fix mentioned in any of the release notes.
I'll test again, but last time I tried the latest version the issue still existed, or at least I could not complete the required selections for SBUS with SPORT telem using a TXMOD

PR #5689 (2024-12-12): feat: S.Port telemetry for PPM/SBUS external module
Fixes #5064

Summary of changes:
- add S.PORT telemetry to PPM external modules
- add S.PORT telemetry to SBUS external modules