EdgeTX/edgetx #5066: Suddenly couple of sound tracks go bad...mispronounced in English

ror1948, 2024-05-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

setup on new Boxer for simple plane
I used sound track "armed" and "disarmed" on SF switch
at first it worked.
Then did play background music on S3-2 and that worked
now  sound track "armed" and "disarmed" sounded like  "at" and "disat"
replaced files on sdcard with saved backup files...same result

### Expected Behavior

sound track "armed" and "disarmed" should sound as shown

### Steps To Reproduce

setup on new Boxer for simple plane
create new basic 4ch model
-press MDL 
-go to special functions
-program SF up for sound track "armed"
-program SF down for sound track "disarmed"
-program S3-2 BGmusic  to play Aliz20.wav
for some reason S3-1 turns the music off without being programed
[paul.zip](https://github.com/EdgeTX/edgetx/files/15440867/paul.zip)

### Version

2.10.0

### Transmitter

RadioMaster Boxer

KAD370, 2024-05-27:
I get very slow speach of telemetry values. Horus x12s with touch mod

scooper1, 2024-07-23:
version 10.2  voice prompts are stuttering  - seems to  be getting more broken up with newer firmware 
tx16s mk 1
elrs protocol