EdgeTX/edgetx #5069: Global Function Parameters Reset on Saving to/from Tx

mhk1058, 2024-05-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Some global function parameters reset to unallocated in Companion when i) Reading Models and Settings from Radio and ii) When opening a previously saved local .etx file the first time. Set the parameters again and save and they seem to remain.
[image]
[image]

As you can see, the parameter for rersetting timer 1 is persistant.

### Expected Behavior

Parameters are persistant when tarnsferring to and from radio and when saving file locally.

### Steps To Reproduce

Set global functions as per attached image on radio in companion.
Save file locally
Write to radio
Disconnect radio
Reconnect 
Read from radio - check parameters... usually gone.
Reset parameters, save file locally and write back to radio
Disconnect radio
Open local file with radio disconnected.. parameters likely gone again.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

3djc, 2024-05-25:
Is your radio inputs properly configured? Check in SYS/Hardware

mhk1058, 2024-05-25:
> Is your radio inputs properly configured? Check in SYS/Hardware

Do you mean here?

[image]

I think so... this looks OK to me.

elecpower, 2024-05-25:
> Some global function parameters reset to unallocated in Companion when i) Reading Models and Settings from Radio and ii) When opening a previously saved local .etx file the first time. Set the parameters again and save and they seem to remain.

So appears a conversion issue.

On the first Read Models and Settings or opening an existing etx are the hardware settings as per your screenshot?

mhk1058, 2024-05-26:
> So appears a conversion issue.
>
> On the first Read Models and Settings or opening an existing etx are the hardware settings as per your screenshot?

Hi, Yes, that's it. Easiest way to replicate it is to:
i) Set up the global functions as per the first image, 
ii) Make sure they are working on the Tx
iii) In companion, read models and settings from radio
iv) Parameters then as per second image

I think the opening of the local saved file issue my be me saving the file from the radio... I'm now struggling to replicate that if I reset the parameters before saving.

pfeerick, 2024-05-26:
I can reproduce this exact sequence... it just seems like for some reason CPN can't read the GF pot settings in the file properly initially, but after it has saved it, it seems fine... radio can read it, and CPN *seems* to accept the file again after the radio has re-written it. 

[tx16s_5069.zip](https://github.com/EdgeTX/edgetx/files/15446845/tx16s_5069.zip)

elecpower, 2024-05-26:
#4768 solved models but broke radio as they share the same yaml decoding method

PR #5072 (2024-05-27): fix(cpn): global functions yaml decode
Fixes #5069
closed 2024-05-27 by pfeerick