EdgeTX/edgetx #5070: Contrast not working properly in 2.10.0 on Radiomaster Pocket

keithluneau, 2024-05-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I can change the contrast value and it seems to work as expected, until a power cycle. After turning the radio off and back on, the contrast appears to work at the default value until I go into the settings and scroll down to highlight the contrast setting. At that point the contrast changes to the value set. I will include a link to a video showing the issue.

https://www.youtube.com/watch?v=NBwMdTod_w8

### Expected Behavior

The set contrast value should be active after a power cycle without needing to enter the settings and select the contrast setting line before it becomes active.

### Steps To Reproduce

See video above.

### Version

2.10.0

### Transmitter

RadioMaster Pocket

3djc, 2024-05-25:
I just tried it here, and I cannot reproduce it. Could you post your RADIO and MODEL folder ?

keithluneau, 2024-05-25:
[pocket.zip](https://github.com/EdgeTX/edgetx/files/15445005/pocket.zip)

Sure thing!

keithluneau, 2024-05-25:
For what it's worth, I downloaded the SD contents and replaced radio.yml, and the problem seems to have gone away. I placed my old radio.yml file back on the card, and the problem is back. I'm not sure how, but it seems the issue is contained there... ;) This was a stock radio fresh out of the box that I updated to 2.10.0, then let the Chrome based tool update the SD contents as usual. I then used the latest Companion to download models and radio settings, removed the stock models, and created one new model. I wrote those back to the radio. I'll repeat those steps again and see if something gets corrupted in the process and report back.

keithluneau, 2024-05-25:
Before I got that far, I went through the settings and change a few things to my liking, and the problem came back again. It seems like something is getting messed up in radio.yml and causing it.

keithluneau, 2024-05-25:
I'm pretty confident I've figured out the issue now, and it's admittedly something I should have checked earlier on... lol

I replaced the SD card with a known good one, and can't reproduce the problem again. Just to be thorough, I even mirrored this known good, working card over to the SD card that came in the new radio, and the problem immediately returned. I put the good card back in and no more problems, everything works as expected!

So, the questionable quality of the included SD card strikes again... lol With that, I'll close this.

Thanks for heading me off in the right direction!

closed 2024-05-25 by keithluneau

markusressel, 2025-02-26:
Thx, I had the same issue.
I bought a cheap "Samsung 32GB Pro Endurance" Micro-SD card, copied everything over, and it magically all started to work.