EdgeTX/edgetx #5074: NV14, In model setup "Open Channel Mon." shortcut is not reacting on touch

juricabi, 2024-05-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After upgrade from 2.8.5 on NV14 radio to 2.10.0, new shortcut for channel monitor is not reacting to touch.  On some tabs it gets highlighted, on some not, but does not opet channel monitor on any tab.  Opening 
 channel monitor from main menu works normally. 

[image]

https://github.com/EdgeTX/edgetx/assets/29048539/1b9a2948-30a1-445b-b837-9e806db904ce

[image]

### Expected Behavior

It is expected that channel monitor tab is opened with shortcut.

### Steps To Reproduce

1. Go into Model setup menu
2. Touch Open Channel Mon. shortcut
3. result - screen of channel monitor is not opening

### Version

2.10.0

### Transmitter

Flysky NV14

pfeerick, 2024-05-29:
Because that button is so small, it is near impossible to press it anyway, so while this is a bug (in that the button doesn't do anything when you do manage to press it and not the top tab bar), I we need to rethink that one... mixes can have their own channel bars now, and there is the "hats mode" that was introduced, which lets you use the trim hats as navigation keys, so pulling the left trim hat down when in navigation mode would get you to the channel monitor also. So it can either be removed and be done with it, or reimplemented in a way that is more usable.

pfeerick, 2024-07-06:
Button has been made a little bigger, and more likely to be pressable. Fix will be present in 2.11

closed 2024-07-06 by pfeerick