EdgeTX/edgetx #5085: Stick, Pot, and Trim names are missing from pop-up lists (TX16s Mk2 2.10)

HippoDan, 2024-05-29:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When selecting an input, a pot, or a trim switch, many of the items appear in the list with no names, and are impossible to tell apart.  For some reason T5 and T6 still have names.

### Expected Behavior

Selectable lists of inputs should show either the names given to them, or the default names.  (Trim switches don't seem to be nameable.)

### Steps To Reproduce

In the model editing pages, choose a source for throttle, or a trim switch for same.  Same problem on mixer pages.

### Version

2.10.0

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

macOS

### Anything else?

[image]
[image]
[image]

pfeerick, 2024-05-29:
Check your Radio Settings -> Hardware -> Pots configuration - the names are probably spaces `   ` instead of blank `---`.

HippoDan, 2024-05-29:
Thanks, that was it. I see the trims took the names of the sticks they were near. Is there any way to get them to show up with their correct numbers?

Either way, I think this problem can be marked as solved.

pfeerick, 2024-05-29:
I don't think so, no. The four next to the sticks take on the name of the axis/stick they normally configure, and the remainder keep their stock/factory numbering since they don't belong to any particular axis.

closed 2024-05-29 by pfeerick