EdgeTX/edgetx #5087: RadioMaster MT12 locks up after modifying a logical switch.

inventor7777, 2024-05-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After a logical switch is modified, the radio freezes until the RTN button is pressed. 

In addition, I think (I can’t confirm due to the absence of a simulator and only having one main radio) you now only click once on the switch to edit or copy, as holding no longer seems to work.

### Expected Behavior

The radio should not freeze after modification of a LS.

### Steps To Reproduce

Open logical switches and delete one. The radio won’t respond until you click the RTN button.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Anything else?

I’m running yesterday’s (5/29/24) nightly build on my RM MT12.

3djc, 2024-05-30:
Confirmed, exist on simu too, likely navigation issue

3djc, 2024-05-30:
And now I cannot replicate anymore. If you can replicate, could you make a video ?

inventor7777, 2024-05-30:
@3djc here is a video. I am able to replicate it every time. I’m traveling right now, so if you need further information I won’t have any until next week. 

https://github.com/EdgeTX/edgetx/assets/142270525/fd809baf-0d50-4be4-8267-e29df2a78cbe

inventor7777, 2024-05-30:
Interesting how it only replicated for you for a brief period.

3djc, 2024-05-30:
No, got it, but it needs a specific sequence. It was introduced by https://github.com/EdgeTX/edgetx/pull/4849

What i do to reproduce, navigate to an ls, scroll in the ls menu, get out out menu, issue

pfeerick, 2024-05-31:
> In addition, I think (I can’t confirm due to the absence of an MT12 simulator and only having one main radio) you now only click once on the switch to edit or copy, as holding no longer seems to work.

Correct... there is no other action anyway so questionable if press and hold is needed to bring up the menu.

closed 2024-05-31 by pfeerick

inventor7777, 2024-05-31:
> Perhaps it should be press and hold for the menu for consistency though?

I sure think so. I was a bit puzzled by it at first, since all the other menus are press-and-hold.

PR #5090 (2024-05-31): fix(bw): unresponsive rotary encoder after using popup menu on LS page
Fixes #5087

Side effect of ENTER key handling arising from changes in #4849