EdgeTX/edgetx #5088: When changing mixes (switched) there is a brief ghost input.

inventor7777, 2024-05-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a launch control mix that activates when the rear button is held down for 0.5 seconds. However, occasionally and quite randomly, the channel gives a ghost input for about 0.1 seconds. The value seems to be about -40 for the brief second it’s onscreen in the Channel Monitor. 

### Expected Behavior

There should be zero input while a mix is switching.

### Steps To Reproduce

My current setup works like this:

L01: when rear button is pressed for more than 0.5 seconds (active for 10 seconds)
L02: when throttle is pushed past 40 more than 0.8 seconds (active for 6.5 seconds)
L03: when L01 is 100 and !L02 is active 
(Basically logic to activate the launch mix when the rear button is pressed for 0.5sec, but disable the mix once the launch has occurred)

So the mix is active when L03 is active.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Anything else?

The issue only seems to occur when initially activating the second mix. It does not seem to occur after the second mix deactivates. I would record a video, but I’m traveling and I hope someone else can reproduce the issue. If not I’ll send a video when I get back.

inventor7777, 2024-05-30:
I did forget to mention that L01 has a PlayTRK function “launch control” activated and there is a haptic every two seconds (connected to L03)

3djc, 2024-05-30:
We will need you model file for that, but usually that can happen if you did not set catchall mix line

inventor7777, 2024-05-30:
What is a catchall mix line? @3djc

inventor7777, 2024-05-30:
> We will need you model file for that, but usually that can happen if you did not set catchall mix line

I’ll send model files as well when I get back.

pfeerick, 2024-05-30:
I believe JC is referring to having a default mix line... I e. If you have
a mix controlled by a switch, you really also need a "no switch" mix line
to cover that possible split second when between states as logic conditions
still being evaluated, etc.

inventor7777, 2024-05-30:
Makes sense. However, one of the mixes already is that: it’s not switched at all and is by default active.

inventor7777, 2024-06-03:
@3djc @pfeerick 
The issue persists with the latest beta. Here is the model.yml file and a video showing the problem. (the problem seems quite intermittent and random (happens about once every five times).

My ADC filter is on, and the receiver wasn't even connected (it does still happen when it's connected, which causes the car to jump). My radio is the 4-in-1 Multi-module.

[model01.yml.zip](https://github.com/user-attachments/files/15537476/model01.yml.zip)

https://github.com/EdgeTX/edgetx/assets/142270525/1f9db534-85cc-4a0e-a518-38c3ff567840

philmoz, 2024-06-03:
Try setting 'Slow dn' to 0 on the second throttle mix line (LACTRL).

I can reproduce this in the simulator (on other radio types) - it may be a bug or it may be by design.
That part of the code is extremely hard to understand.

inventor7777, 2024-06-05:
> Try setting 'Slow dn' to 0 on the second throttle mix line (LACTRL).
Thanks! I think that worked. Why does that affect it though?

philmoz, 2024-06-05:
> > Try setting 'Slow dn' to 0 on the second throttle mix line (LACTRL).
> 
> Thanks! I think that worked. Why does that affect it though?

I think it has to do with the way the slow up / down code works on mix lines that can be turned on and off with a switch.
It looks like an initialisation problem when the mix line re-activates after a previous de-activation.
Unfortunately the code is extremely complex and difficult to understand and I'm reluctant to tinker with it at this point.

inventor7777, 2024-06-05:
> It looks like an initialisation problem when the mix line re-activates after a previous de-activation

Ah, ok.

> Unfortunately the code is extremely complex and difficult to understand and I'm reluctant to tinker with it at this point

Hm. That’s unfortunate, but makes sense. Hopefully it can eventually be fixed and the code cleaned up a little (if possible)

inventor7777, 2024-08-05:
@philmoz 
@pfeerick 
@3djc 

This issue is getting more and more serious due to the way I am using the mix.

Today I went to use launch control on my car after updating to a later version of 2.10 and the issue happened again. However, this time it was far more serious as the car did a standing front flip, shot backwards, diffed out, blew out a driveshaft and tire, before slamming into the curb and shutting off. Luckily the only damage was to the driveshaft, as i was able to reglue the tire. Thank goodness there were no people nearby (except me).

Is there any way this issue can be escalated a bit in hopes of fixing this bug? Even if it comes in the form of an indirect fix (like maybe a switch to ignore negative values? (I tried adjusting weight & offset, but then the expo and delay work very strangely))
I totally understand about it being hard to fix, but it’s certainly fixable or at the very least, have some form of switch to prevent such errors in future.

Thank you!

(For clarification, the reason the car stayed in reverse long enough is because I have a delay up set in the mix, and some expo, so the car stayed in full reverse for a full 4 seconds. I am so glad it was not one of my much higher powered vehicles,)

philmoz, 2024-08-05:
Do you still have mixer lines with both a switch and slow up or slow down set?
If so, you will need to find a way to avoid this - slow up and down do not work correctly on mixer lines that are also controlled by switches.

Without rewriting the mixer there is no simple solution to this that I know of.

If you want to share the model file I can take a look and see if there is way to achieve the desired result while avoiding this conflict.

pfeerick, 2024-08-06:
i.e. maybe you can use the fade in/out options in the flight/drive mode somehow in conjunction?

3djc, 2024-08-06:
@philmoz the model file is here https://github.com/EdgeTX/edgetx/issues/5088#issuecomment-2145755290 (haven't had the time to look at it yet)

philmoz, 2024-08-06:
@inventor7777 

I'm not sure I completely understand the model setup (using the model01.yml posted earlier).
Anyway here's a version that moves the L12 switch control from mixer lines to input lines.

[model08.yml.txt](https://github.com/user-attachments/files/16508202/model08.yml.txt)

inventor7777, 2024-08-06:
Thanks very much! When I get free time later today I will review the model and see if it works.

inventor7777, 2024-08-06:
@philmoz 

Unfortunately it does not work for my purposes (I don't think) as I cannot apply the correct expo type. The standard launch control is slow up until about 45% through the range, then it floors the car for max acceleration. With the new setup it does indeed have a slowdown set, but applying expo does not affect anything as far as I could tell. 

The only solution without rewriting the mix code imo is just to add a simple switch under the mixer settings to "ignore negative values" or similar. I truly hope some form of solution can be found soon.

inventor7777, 2024-08-07:
Ok, so I made a logical switch that activates when throttle is at ~0 and while this is active it holds the throttle at 0. So far it seems to be working fine. Haven’t tested with my cars yet but the Channel Monitor looks promising. Hopefully the 1/10 sec delay of the LS won’t be noticeable.

I still hope this can be fixed in later versions of Edge but for now I think these workarounds will work.

inventor7777, 2024-08-07:
I ended up updating it so that the logic goes like this:

L01 - when launch button pressed -active for 8 sec
L02 - when lo2 is active - delay 0.1 sec (this is the input that turns on the launch mixer)
L03 - when L01 is active & throttle at 0 - delay 0.2 seconds (so there’s no unnecessary delay if I have to slam on brakes)

so that the overriding of the throttle only happens in launch mode and I delayed the activation of the mixer activate so that the ghost input doesn’t happen.

inventor7777, 2024-08-20:
Hi guys. Sorry if I’m being annoying by bringing this up all the time, but now this bug has had some semi-serious consequences. 

I had just swapped my Maxx to 6s and I was tinkering with the launch control to get it right again and I started with a bit of a longer delay. Even though I had my safety buffer in place to hold the throttle at zero it somehow still did the same blip on the throttle channel. 

This time I got extremely lucky not to hit anyone or anything (this was in a residential area just for my tests) it ended up blowing out my $60 belted tires and it could have caused some extreme damage. 

**I say all that to say that I no longer can trust my radio to reliably control my cars without a chance of serious bugs that can cause damage or seriously hurt someone.** I have checked and rechecked my logic and I’m 100% sure it’s a bug and not anything I did. 

I know this requires a full rewrite of the mixers, but at this point I just want this fixed so I can use the advertised features of my radio without worrying about random bugs that will cause the car to run away at full speed. At the very least can you guys start to work on that? @pfeerick @philmoz

philmoz, 2024-08-20:
Are you still using Mix lines with both Slow up/down and a switch configured? 
If so you will have to find a way to remove these from your setup.

pfeerick, 2024-08-20:
A re-write of the mixer is not likely in the short term. If you are using
slow and a switch in this way, you need to stop doing that. How is your
"safety button" working, via replace mix line or channel override SF?

inventor7777, 2024-08-20:
@philmoz Its just Slow Up, no Slow Dn

@pfeerick the safety button is using an override SF, and I’ve adjusted the delays a bit and it does indeed work reliably now without error. However it’s still annoying to me that I have to deal with that.

pfeerick, 2024-08-20:
I'm not saying that it isn't, and that there isn't a bug... simply it is not on the cards. The mixer is well overdue overhaul, but it is too big an ask at the present time. This is something that will affect **all** users, and thus will need a _lot_ of testing before it can even be contemplated being offered in a PR, let alone let loose in a nightly. We should probably also go back to the drawing board on that, and see if there is anything missing from the mixer, since it will be at the complete re-write stage.

philmoz, 2024-08-20:
> Its just Slow Up, no Slow Dn

If the mix line has a switch then you can’t use any combination of slow up or down.

inventor7777, 2024-08-20:
Okay. Unfortunately I have to use both a switch and Slow Up for what I’m trying to do here, thus my annoyance. I did give your other solution another try but I simply cannot get it to work for what I want.

inventor7777, 2024-08-20:
I am open to any suggestions to get around this issue. In addition, now that I need longer delays for launch control due to 6s swapping my car I’ve now encountered issues with limitations shown in #5118. I am really wondering now if it is even possible to do what I want in the current version of EdgeTX.

 I just want a launch control function that enables when a button is pressed and disables after launch with full control restored if launch is aborted (currently the car will slam in reverse after braking due to the slow up function). At the moment I don’t think that’s possible at all without encountering either bugs or missing features. I’ve noticed that the new Spektrum IXSR has launch control built in, which sounds quite intriguing. I wonder if something similar could be implemented eventually?
 
[image]

inventor7777, 2024-08-20:
@pfeerick I get where you’re coming from! I simply wish that there was a way to do what I’ve outlined in my comment above, but I suppose it’s simply not possible ATM.

pfeerick, 2024-08-20:
This *may* be where you need to go into the deep end that is Lua mixer scripts... I'm not sure at the present time you can actually manipulate the mixer into doing what you are actually  trying to do (if I'm understanding this right), which is change that thump of the throttle stick into a smooth linear curve, which is also dampened to arbitrary limits so it doesn't spin out. Although... it's _almost_ like you want a custom curve for short duration, when the throttle transitions from 0 to 100 :thinking: ... which has me wondering about using a logical switch to detect the throttle press, another as a timer, link the two, and use that as the switch that determines which mix line (and thus which throttle curve) to use...

3djc, 2024-08-21:
a LUA mixer script certainly could do it, but thats a gui less experience, not great

pfeerick, 2024-08-21:
Indeed... so surface specific mix options when? lol At least you can pass
values into mixer scripts, so it wouldn't be completely unwieldy and like
pulling teeth...