EdgeTX/edgetx #5099: Widget focus issue after selecting layout in telem config screen

shellixyz, 2024-06-01:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

After selecting a layout in the telem config screen it is not possible to focus the toggles (top bar, flight mode, ...). One needs to switch page or the telem config screen and come back to be able to access them

### Expected Behavior

The toggles need to be focusable / accessible after changing the layout type

### Steps To Reproduce

1. Go to telem config screen
2. Change layout
3. Try to focus the "top bar" toggle or any other of the toggles in the page

### Version

2.10.1

### Transmitter

Jumper T16

pfeerick, 2025-02-14:
> After selecting a layout in the telem config screen it is not possible to focus the toggles 

Reproduceable, when using key navigation only. I think this may have been an issue that was fixed in 2.11 but still present in 2.10. You actually can get to the toggles, it is just the focus order that is wrong... it will jump to the button at the bottom on the first step, and then jump back up to the first toggle.

pfeerick, 2025-05-15:
Appears to actually be fixed in 2.10.6 (was present in 2.10.1), but can't immediately see what changed fixed it.

closed 2025-05-15 by pfeerick

philmoz, 2025-05-15:
> Appears to actually be fixed in 2.10.6 (was present in 2.10.1), but can't immediately see what changed fixed it.

Still present in 2.10.6 - you need to have multiple views, only shows on views with the 'Remove' button at the bottom.

I'm not inclined to fix it though, unless you really think it is necessary.

pfeerick, 2025-05-15:
Ah, ok, thanks... and no, me neither, as I couldn't readily see the fix you did for it that went into 2.11, and it is only focus order, not complete inability to access those options as originally reported.