EdgeTX/edgetx #5106: Boxer and Zorro: Missing spaces, letters are cancelled in certain menus

bernd25, 2024-06-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

In some menus e.g. in Logical switches I add the following parameters:  
a=x TPWR  7mW
Back in the list it looks like this "a=x TPWF7mW", whereby the 'R' is no longer recognizable as such.  It could also be an 'F'.
In other cases a letter disappear completely. 

[image]

### Expected Behavior

In the LS list it should look like:
a=x TPWR 7mW
as in older versions prior v. 2.30.0 

### Steps To Reproduce

Add a logical switch with same parameters and switch back to the list.

### Version

2.10.1

### Transmitter

RadioMaster Boxer

### Anything else?

Due to the lack of spaces, some menus are almost unreadable. 
This applies equally to RM Boxer and Zorro.

philmoz, 2024-06-02:
2.9 did not show the source type symbol at the start of the name.
[image]

Due to the limited space we should probably remove the source type symbol to avoid overlap.

pfeerick, 2024-06-02:
Indeed... only enough space for 4 characters there...

PR #5109 (2024-06-03): fix(bw128): telemetry source name overlaps V1 value on logical switch list view
Fixes #5106 

Don't display the telemetry symbol due to limited space available.
closed 2024-06-03 by pfeerick