EdgeTX/edgetx #5110: Internal MPM will not flash newest firmware. Jumper T Pro V2

Scoot49, 2024-06-02:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I flashed to the latest EdgeTx version v2.10.1 Centurion.  As recommended, I tried to update to the latest internal MPM firmware mm-stm-serial-aetr-air-v1.3.4.0. immediately after.     I continually get an error message stating "FW update error - Device not responding".    I am entering into the SD Card FIRMWARE folder on the transmitter, long pressing on the the "mm-stm-serial-aetr-air-v1.3.4.0." file and selecting "Flash Int. Multi".      This is when I get the error message.     I cannot get FLASH Multi app to cooperate either.  It appears to connect correctly, but when I try to read the Module it delivers a "FAILED TO READ THE MULTI-MODULE" error.   In the past I have successfully flashed and updated several EdgeTx versions, as well as MPM firmwares.  I am unsure of what is going on here. 

### Expected Behavior

The MPM should flash correctly and update the internal MPM to "mm-stm-serial-aetr-air-v1.3.4.0. ".

### Steps To Reproduce

Enter into transmitter settings.
Navigate to page 2 **_SD CARD_**
Open **_Firmware_** folder
Select "**_mm-stm-serial-aetr-air-v1.3.4.0._** file, long press until selections pop up.
Select  **_Flash Int. Multi_**.  

Within 2 seconds error  message "**_FW update error - Device not responding_**" is displayed 

### Version

2.10.1

### Transmitter

Jumper T-Pro V2

### Operating System (OS)

Windows

Dingobrain, 2024-06-03:
I have exactly the same issue with my Jumper T-pro V2 and internal JP4in1 Multi module.
Tried updating from the SD card: "FW update error - Device not responding". 
Tried with FLASH Multi: "FAILED TO READ THE MULTI-MODULE"

3djc, 2024-06-03:
Unfortunately I don't have one to tr6 to replicate

3djc, 2024-06-03:
The closest I have is a Zorro with int mpm, that works fine updating

3djc, 2024-06-03:
Even closer, I found a Tpro (non v2), that uses the very same serial port as V2, that works fine

Scoot49, 2024-06-04:
That does make sense Dingobrain, maybe for your circumstance I guess.  I didn't check the spectrum analyzer for output, however, I was able to bind and fly several aircraft using the current MPM firmware that is  loaded as I was unsuccessful in updating.  I did backup my SD cards contents before I started any of the Edgetx and MPM updates, so I am going to drop the 1.3.4.0 firmware into that backup file, and try to flash that way.  I will report back tomorrow with the results.

Scoot49, 2024-06-04:
> Is the module turned on for this model ? What is displayed in status line ? And you confirm you have the model with Oled screen (text is white on black background) ?

The module is on as I was able to fly several aircraft using the Centurion update and the un-updated MPM firmware.  I owned the  V1 TPRO and V2 TPRO but have since sold the V1.  Appreciate where your head is though.

pfeerick, 2024-06-04:
@3djc Can you try with an external MPM (I tried using iRange iRX4) and TPro or v2... I _can_ get both of those to fail to firmware update with 2.10.1, but will update with 2.9.4. Device reset seems to take a little longer than I think it normally should, and then it fails with device not responding after a small movement of the progress bar.

3djc, 2024-06-04:
I'll check if I have an external mpm of nano size

3djc, 2024-06-04:
Unfortunatly I don't have one of that size :(

3djc, 2024-06-04:
Ok, was able to do a frankenradio taking the mpm  module from a v1 to a v2, and I can reproduce, so I should be able to investigate

3djc, 2024-06-04:
Could you try with this firmware ?

[TProV2-TEST-MPM-FLASH.zip](https://github.com/user-attachments/files/15544635/TProV2-TEST-MPM-FLASH.zip)

Scoot49, 2024-06-04:
> Ok, was able to do a frankenradio taking the mpm module from a v1 to a v2, and I can reproduce, so I should be able to investigate

@3djc    So you guys are able to reproduce this and it's just not me updating incorrectly?

  If this helps, I did notice the EEPROM folder is no longer listed on the SD card.    I don't have that laptop or transmitter in front of me at the moment as I'm at work, but I do remember noticing the EEPROM folder is no longer listed.  I cannot remember if the back up I made , (which would have incorporated a 2.9 EdgeTx version) listed the EEPROM folder.  I think I remember reading that the EEPROM may have gone away after a particular update, like Edge 2.8 or something, so I may just be rambling,  in which case ignore.  Just though I'd disclose that in case it helps.  I have updated the Edgetx versions faithfully as released, and maybe I just didn't notice the EEPROM bow out.

3djc, 2024-06-04:
EEPROM folder is totally unrelated, and was a legacy item from Otx time, that never had any purpose in EdgeTx

pfeerick, 2024-06-04:
> was a legacy item from Otx time, that never had any purpose in EdgeTx

Since EdgeTX v2.6, that is. Both EdgeTX v2.4 and v2.5 used EEPROM still for B&W radios. For radios introduced with or after v2.6, this naturally means EEPROM folder is not relevant.

Scoot49, 2024-06-05:
Copy. Thanks for clarification on the EEPROM.  

@3djc   Regarding "TProV2-TEST-MPM-FLASH.zip "   I loaded this onto my SD card.  When I go select and long press, it only gives me an option to flash bootloader.  Not the internal MPM.   Is this correct and by design?

pfeerick, 2024-06-05:
lol... that's the transmitter firmware you need to flash to test the fix ;) Go into the bootloader (with transmitter turned off, hold horizontal trims inwards and power on), and flash that firmware, and then try flashing the MPM firmware again to see if that fixes the issue for you.

Scoot49, 2024-06-05:
@3djc Got impatient and went for it.   Flashed _TProV2-TEST-MPM-FLASH.zip_ via Bootlloader.  Write was successful.   Restarted transmitter.   Flashed internal MPM with __mm-stm-serial-aetr-air-v1.3.4.0.__   Everything states that it was successful.  **Status lines** are indicating the v1.3.4.0 was accepted.     Anything I need to be aware of during testing??    What was the trick?

3djc, 2024-06-05:
@pfeerick , so I built this poor man TProv2 to JR module adapter :)

[image]

The issue is I can flash perfectly fine my full size MPM  with TPro V2 and 2.10.1 (which reinforces my belief that the internal mpm on those radio is just a tad slower)

pfeerick, 2024-06-05:
Nice! :) As I just commented over in the PR, it's both the iRange iR4Xlite and RM4in1, and the RM4in1 will not work at all with the TProV2 at all, so I think there is more to this puzzle.

3djc, 2024-06-05:
This is the module I test with

[image]

[image]

As you can see, it works fine

[image]

pfeerick, 2024-06-05:
>  I'll struggle to fix something I can't reproduce

What? Why? I'm losing confidence in you JC... I thought you were the B&W radio whisperer. :rofl: :rofl: 

Fully understand... just something to keep in mind in case there are more reports soon ... although if it is limited to radios like the TPro (regardless of version) and (certain?) external MPM, I very much doubt there would be many! :zany_face: Hopefully will be able to look into it further before that happens. :upside_down_face:

3djc, 2024-06-05:
What other nano module do you have ? Elrs or ghost ? Maybe we could find an issue there as well

pfeerick, 2024-06-05:
MPM (2 of), ELRS (3 of I think - RM, Jumper and BetaFPV), XJTLite. I do have full sized Ghost, R9M+2019, Tracer, Crossfire, ELRS, XJT, so could make up adapter similar to yours and test a bunch of other modules... someday soon :tm:

PR #5124 (2024-06-05): fix: timings for MPM update due to faster F4 MCUs
Use proper timeout instead of MCU speed dependent loops

This fixes #5110

If confirmed solving the issue, needs to be cherry picked on main too