EdgeTX/edgetx #5112: RadioMaster TX16S MK II Max bricked when flashing in DFU mode via EdgeTx Buddy

LTabora, 2024-06-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I proceeded to flash EdgeTX 2.10.1 to my radio running 2.9.4 (Providence) via DFU mode using EdgeTx Buddy. This bricked my radio. 

At first I thought I had chosen the wrong radio firmware, but I attempted to retry the procedure several times (3-5 times, always same results. The process seemed to work flawlessly, without any glitches, yet the radio always resulted bricked). 

I also attempted to do so via the STM32CubeProgrammer App, as I was attempting to fix my radio. (Same thing, process finished successfully, but the radio wouldn't start.

I ended up fixing my radio by flashing 2.9.4 once again. This worked perfectly.

I even attempted to re-flash 2.10.1 after I had it working once again with Providence, to no avail. I had to re-flash Providence to bring it back to life.

### Expected Behavior

The firmware flash procedure to work as expected.

### Steps To Reproduce

1. Connect radio (RadioMaster TX16 MK II Max) to PC in DFU mode
2. Launch EdgeTx Buddy in a chrome brower
3. Select 2.10.1 firmware
4. Select RadioMaster TX16S radio
5. Make sure to select proper STM32 Bootloader device
6. Click Flash

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### Operating System (OS)

Windows

### OS Version

Windows 11 Pro

3djc, 2024-06-03:
Just tried it on 3 TX16S, all working fine, so it is a local issue. (You did not mention a language so I have assumed Englsih)
-try on another PC (anti virus could be playing dirty for example)
- try another USB cable

LTabora, 2024-06-03:
Yes @3djc, the language is english.

It's a curious issue because it has worked perfectly before, still works great for 2.9.4 (as I've flashed it back to 2.9.4 twice after several failed attempts to upgrade).

The problem isn't the process itself, from what I can tell, but rather something within the radio. The process runs through just fine.

3djc, 2024-06-03:
I think something PC side, not radio side. DFU is built in STM32 MCU, so nothing can happen here. The only thing that can happen is that the firmware is somehow corrupted while been sent, which can be down to usb cable, anti virus,...

pfeerick, 2024-06-04:
I've not had this happen with any of my TX16S's either. 

I have heard of exactly one other report of this happening, and the poster even tried pulling the RTC clock battery with no change in behaviour. But for some reason, for him, flashing OpenTX and then flashing back to EdgeTX 2.7 (as that was working also with no issue) and then v2.10.1 was fine. So I'm wondering if perhaps it is still something stuck in the RTC, and may not have waited long enough for it to clear/reset or some other weirdness going on. 

What exactly do you mean by bricked? That the radio doesn't even turn on? Or do you rather mean it would power on but be stuck at the splash screen? As this is what happened in his case.

LTabora, 2024-06-04:
Hi @pfeerick , 

By bricked I mean that the radio doesn't even start (turn on). When you push the power button, the "chicken leg" buttons flash sequentially from left to right and back, and that's about it. 

I did try pulling the main battery out for a bit and waiting, and that did nothing. I didn't try removing the RTC one.

What I find very odd, is that it works flawlessly, however many times I try flashing it with 2.9.4; which has allowed me to try many times to flash 2.10.1, to determine if I can find the smoking gun.

One thing I do recall, is that when I yet hadn't found out that flashing back to 2.9.4 solved my immediate problem, I also tried flashing 2.10.1 with the STM32CubeProgrammer app (as I was attempting to recover my radio), and in several of those occasions, the validation step failed with the following error: 
data mismatch error 0x08000003 (byte = 0x00 instead of 0x10).

Of course, I panicked even more, but I later found out that reflashing 2.9.4 would always fix it back, so....

philmoz, 2024-06-04:
That sounds like a hardware issue and the firmware is not being written to the flash memory correctly.

I would try flashing the firmware in DFU mode and then reading it back and comparing to the original file.

LTabora, 2024-06-04:
> That sounds like a hardware issue and the firmware is not being written to the flash memory correctly.
> 
> I would try flashing the firmware in DFU mode and then reading it back and comparing to the original file.

The odd thing is that it ONLY happens with 2.10.1; With 2.9.4, everything works perfectly!

philmoz, 2024-06-04:
It could just be that the value that are not being written correctly in 2.10.1 are different in 2.9.4 and do not cause an issue.
I've seen stranger things.

Reading the flashed firmware back and comparing to the original file is the best way to check.

LTabora, 2024-06-04:
According to STM32CubeProgrammer, there are no differences!

(I also tried flashing 2.10.0 with exactly the same results.

[image]

LTabora, 2024-06-04:
Also.... what would a full chip erase operation do? 

[image]

pfeerick, 2024-06-04:
I was just about to suggest that... as I got a verify error also when flashing my normal testing TX16S with STM32CubeProgrammer, so it seems the auto-erase being instructed by the software doing the programming isn't blanking all the sectors being used,  hence the failure to verify. The Full Chip Erase will blank all the sectors properly, and then you should be able to write the firmware again and it should pass verification this time.

LTabora, 2024-06-04:
* UPDATE *

I just tried updating the firmware via the SD card while the radio was on 2.9.4

- Loaded the firmware on the SD card, 
-I launched the bootloader and flashed 2.10.1

This had my radio working on 2.10.1!! 

I then proceeded to upgrade the bootloader from 2.9.4 to 2.10.1; the process completed successfully. I restarted the radio, and once again, it wouldn't power on 🤣.

The issue seems to be within the bootloader part of the code, as a 2.9.4 bootloader runs the 2.10.1 firmware perfectly fine!

LTabora, 2024-06-04:
> I was just about to suggest that... as I got a verify error also when flashing my normal testing TX16S with STM32CubeProgrammer, so it seems the auto-erase being instructed by the software doing the programming isn't blanking all the sectors being used, hence the failure to verify. The Full Chip Erase will blank all the sectors properly, and then you should be able to write the firmware again and it should pass verification this time.
> 
> Note: This didn't affect it's ability to run... 2.10.1 still works fine on this TX16S

Also tried this. No luck either. Same results.

The only way I've made it work, is by doing what I noted above; flashing the 2.10.1 firmware from the SD card while running the 2.9.4 bootloader. This allowed my radio to actually execute and successfully run the 2.10.1 code. 

Once I also upgraded the bootloader, the radio wouldn't boot anymore.

pfeerick, 2024-06-04:
>  2.9.4 bootloader. This allowed my radio to actually execute and successfully run the 2.10.1 code.

With multiple reboots? Very strange! 🤯 😖

LTabora, 2024-06-04:
One thing I'm noticing, as I'm verifying things, is that something happened to my radio (apparently while running 2.10.1) and at least for the current model, as I haven't validated the others:

- Calibration was completely off, had to recalibrate everything
- All play track functions were repeating every 1s 
- I had to go to hardware on the radio and re-enable the sliders, pots and back buttons. 
- I can't seem to select the 6 position switch when trying to assign it in one of my functions.

Was this the SD corruption you were referring to @pfeerick ?

LTabora, 2024-06-04:
> > 2.9.4 bootloader. This allowed my radio to actually execute and successfully run the 2.10.1 code.
> 
> And worked ok with multiple reboots? Very strange! 🤯 😖

Well to be quite honest, I only booted it once after the firmware upgrade and proceeded to upgrade the bootloader right after; so I'm not sure if it would've handled multiple reboots with the 2.9.4 bootloader...

pfeerick, 2024-06-05:
> * Calibration was completely off, had to recalibrate everything
> * I had to go to hardware on the radio and re-enable the sliders, pots and back buttons.
> * I can't seem to select the 6 position switch when trying to assign it in one of my functions.

These three will be inter-related... you *may* need to do stick calibration if you have been switching back and forth between versions, and it seems like the ADC changes somehow managed to knock out the Pots config, so you need to configure sliders, pots and 6POS again. Do the (stick/axis) calibration *after* setting the pots/sliders/6pos again thould, and make sure you calibraate the 6POS also when doing the stick/pots full movement step... i.e. press each button in turn from 2 to 6 with say about 0.5 second delay between presses (the screen _should_ update position) and then 5 back to 1. 

> * All play track functions were repeating every 1s

This could be forward/backward corruption of model settings also

pfeerick, 2024-06-06:
Just as a side note, I encountered a somewhat similar issue with my TX16S recently with a 2.10.1 firmware build, but it was because the firmware download on the SD card was corrupt. While the bootloade flashed and worked, the firmware however flashed and  didn't work... locking up the radio. Replacing the firmware file with a fresh copy and flashing it again via the bootloader worked. STM32CubeProgrammer device memory check also found no differences at this point. 

Not that this helps you as your TX16S doesn't seem to like that bootloader/firmware combo for some reason :(

LTabora, 2024-06-07:
> Not that this helps you as your TX16S doesn't seem to like that bootloader/firmware combo for some reason :(

Thanks for the follow up @pfeerick . It's most definitely an odd thing going on. 

I went ahead and ran calibration following your suggestions and all looks good now, under 2.9.4. 

Somehow, given my latest test, it seems to be something within the firmware. 

I paid more attention, and the response from the radio, when you try to power it on (on 2.10.1), is exactly identical to when you plug the USB cable to put it in DFU mode. The multi-buttons (chicken legs) flash left to right several times and the left one is lit last, until you let go of the power button. Whatever all this means ☺️.

pfeerick, 2024-06-07:
The "chicken legs" are controlled by a separate processor, which always does the animation on power on... so basically it just means it is powering on... and that is about it! :rofl: :confounded:

miguelqtkgas, 2024-06-16:
Exactly the same problem here as LTabora. In my case is a TX16S MKI. It seems to be a firmware 2.10.0/1 bootloader issue.
Bootloaders before v2.10.0 seems to work fine, but since v2.10.0 my TX16S MKI no power on (only "chicken legs").
I also can run firmware 2.10.0/1 with bootloader v2.9.4 (seems to work , at least inside home).
¿Any ideas? Thanks a lot!

LTabora, 2024-06-16:
> Exactly the same problem here as LTabora. In my case is a TX16S MKI. It seems to be a firmware 2.10.0/1 bootloader issue.
>
> Bootloaders before v2.10.0 seems to work fine, but since v2.10.0 my TX16S [...]

I wonder why it isn't affecting more ppl/radios, and rather only a few 🤔 🤷🏻‍♂️?

LTabora, 2024-07-04:
@pfeerick , @miguelqtkgas , I just attempted to upgrade to 2.10.2 but had the exact unfortunate results. Immediately rolled back to 2.9.4

miguelqtkgas, 2024-07-04:
Yeah! Me too. Same behavior as last versions... Returning to v2.9.4

pascalloher, 2024-08-16:
Exactly the same problem here. 2.9.4 works fine, as soon as one of the 2.10.X bootloaders is installed, the 2.9.4 version no longer starts. Bootloader installed via SC Card Screen (Selected EdgeTX firmware file and then selected the option to “Flash bootloader”).

miguelqtkgas, 2024-08-26:
No luck with version 2.10.3, same problem as always. It seems that we will never be able to update our transmitters with new firmware versions... :'(

3djc, 2024-08-26:
Do you have a receiver or gps soldered/connected to a serial port ?

pascalloher, 2024-08-26:
Not a single modification has been made to my remote control, it is in its as-delivered condition. 2.10.3 could be installed with the 2.9.4 bootloader without any problems and seems to run flawlessly. For me, it currently works, it only becomes problematic when the bootloader has to be updated.

LTabora, 2024-08-26:
> No luck with version 2.10.3, same problem as always. It seems that we will never be able to update our transmitters with new firmware versions... :'(

I'm starting to suspect the same. I wonder if RadioMaster will do anything about this particular? It's either a very peculiar piece of code, or we have faulty hardware 🤷🏻‍♂️

miguelqtkgas, 2024-08-26:
As @pascalloher, any hardware modification has been performed to my radiomaster MKI. The only "extraordinary" thing that happened in the past to my TX16S is that during the migration from opentx to edgetx, it was bricked without access to DFU mode. I needed to reset DFU mode following the instructions posted on [https://www.youtube.com/watch?v=ttKKreqHvJc](url). Before this, all updates of edgetx works fine without any problem (until 2.10.x). I post this to look for any coincidence that can help us to find the problem. 
@LTabora, @pascalloher, @pfeerick  ¿Have you ever bricked your TX16S without DFU access in the past and you need to reset DFU?

richardclli, 2024-08-26:
DFU flashing just flash the bootloader and the firmware at the same time.  It may brick the radio if the bootloader has problem.  I tried to flash my TX16S mk1 with 2.10.3 via DFU (bt and fw both updated) and it boots up without any problem.

3djc, 2024-08-27:
Not sure if that thread should be kept opened, clearly EdgeTX is not at fault. It could be hardware issue with the radio, or something around the flashing computer, but nothing requiring an EdgeTX fix

LTabora, 2024-08-27:
I went ahead and reached out to RadioMaster. I haven't tried it just yet, but this was their response:

"Hi Lennin Tabora,

I think some chip-level Option Bytes may be ignored, please refer to the OB Settings in the screenshot, perhaps an Option byte affected the bootloader of 2.10.x, resulting in failure to start normally.

You can also remove the button battery from the motherboard to avoid bad data in the RTC Backup Ram affecting the startup process.

Kind regards,

Radiomaster support team"

[image]

LTabora, 2024-08-27:
> Not sure if that thread should be kept opened, clearly EdgeTX is not at fault. It could be hardware issue with the radio, or something around the flashing computer, but nothing requiring an EdgeTX fix

Wouldn't we want to be absolutely 100% sure before closing it?

At least knowing what the root cause is in case someone else faces the problem?

LTabora, 2024-08-27:
@3djc , @pfeerick , @philmoz ,  @pascalloher , @richardclli , @miguelqtkgas 

I have GOOD NEWS!

I believe that with the help of RadioMaster support (something we should've done a while ago), we've found the culprit.

I connected the radio to STM32CubeProgrammer to validate the Option Byte settings RadioMaster shared with me against what was set in my radio.

I found that the option BFB2 (under User Configuration) was checked in my radio. I went ahead and unchecked it and hit apply.

I then connected the radio to EdgeTX Buddy in DFU mode and retried upgrading my radio to 2.10.3 "Centurion", crossed my fingers, and VOILA!!! The radio upgraded and booted correctly this time!!!

I believe we can now finally put this thread to rest @3djc !!

@miguelqtkgas , I recommend you try this out 👍 ! 

[image]

miguelqtkgas, 2024-08-27:
I have no words... It works also for me! 👍  Thank you very much @LTabora !!! The flag was also checked in my TX16S.
Do you think this info should be in the FAQ or somewhere anyone can find it easily?
Maybe more people are in the same situation and could be very useful this "trick"
Thanks again @LTabora !!! :D

pfeerick, 2024-08-28:
Please leave this open... documentation does need to be updated as this has affected several users now.

3djc, 2024-08-28:
I would personally think the bit flipping is more likely to be the result of an attempt to flash a wrong device, like FC while radio in DFU, something along those line

pfeerick, 2024-08-28:
Risto mentioned an STM32 errata/note about this, so I'm not sure about that... I don't think we (RC hobby) are big enough fish to get a mention from them :laughing: 

Anyway, I've put the start of a "post update troubleshooting" page up now, so we've got something to point people to and add to now. :)
https://manual.edgetx.org/installing-and-updating-edgetx/having-trouble-after-updating#i-updated-edgetx-and-now-my-radio-doesnt-turn-on-or-acts-strangely

closed 2024-08-28 by pfeerick

rcweebles, 2025-01-29:
I just flashed using edge tx buddy on an M1 Mac and bricked by tx16s - even though status screen showed success. Unchecking option BFB2 resulted in the radio working as expected with the latest version of edge tx. Interestingly insert the SD card into a 5 year old TX16S worked prior to changing the BFB2 selection.

Weird.

Should not have skipped the step of replicating the SD card contents prior to upgrading....

hoopermeister, 2026-01-11:
I registered for an account specifically to thank you all for this thread. 
As a very new person to the world of FPV and controllers etc, and not being particularly tech savvy this thread was an absolute godsent. 
Bricked my controller exactly the same way as the OP. 
6 hours following videos and tutorials, and finally stumbled across this thread. 
Solved it immediately.