EdgeTX/edgetx #5118: New type of delay.

nikpoz, 2024-06-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

This is a new feature request, not related with a bug or issue.

### Describe the solution you'd like

While the MT12 has an delay up and delay down feature,  it would be great if another kind of delay could be added. 
The current dalay controls the possitive or negative values related on mid possition. When a delay up for example is set, then for the channel values above mid the delay is used.
The kind of delay I would like to maybe see in the future works like this:
"Speed input" : sets  the desired value when some user input is present on the channel. Use this value in both possitive and negative values related to channel midpoint. Speed input should effect how quick the user input is passed, for example to a servo, altering the seed servo moves to completethe move requested.
"Speed return" : sets  a speed on the channel, during the "return to mid"move.  When user demands no other input, for example from a stick, servo should use this value while returning to neutral possition. 
This is a feature that can be set on many programmable servos, named usually "dumping" and it would just be great if we could expoit its advantages.

pagrey, 2024-06-03:
Maybe I don't understand what you are trying to do but the radio doesn't know when you give the stick input, for example flaps on the throttle channel +100 might be "no user input." For the other channels the springs normally hold them about zero but the radio just knows the position.

If you want to change the speed after the stick is what you define as no stick input just use a logical switch to define that stick position and then add a mix with a different delay based on that stick position. You can add any number of mix lines to change the delay or speed based on the position of the control.

For example L01 true if the absolute value of Ail is greater than zero. Create a mix that sets the channel slow or 3s when L01 is true and slow 0s when L01 is false. When you are moving the stick it is slow, when you let it go or move to center it moves fast.

It seems like a bit of work with logical switches and mixes will get it done just about any way you want.

nikpoz, 2024-06-03:
Sounds good! I'll give it a go and let you know if achieved what I needed. Thanks a  lot!!

inventor7777, 2024-06-03:
Isn’t this already present when combining delay and slow modes? (I also have an MT12)

pagrey, 2024-06-03:
The request involves the rate (delay/slow) changing depending on the control input which isn't just delay and slow. As said, it is a bit cumbersome. Not bad on companion, hopefully that's up and running for the MT12 soon.

inventor7777, 2024-06-03:
Oh, that makes more sense. So basically what you’re saying is that the value of a channel (e.g throttle) can be used as a delay? If so, I also want this feature.

pagrey, 2024-06-03:
EdgeTX allows you to do just about anything which can be great and also difficult to learn. The [EdgeTX User Manual](https://manual.edgetx.org/) is a good place to start and the [EdgeTX Clinic](https://rc-soar.com/edgetx/basics/index.php) is also a great place to learn about how inputs, mixes and outputs work.

MicheleVilla, 2024-06-04:
Nikpoz perhaps meant this:
the rest position of the stick is at the center, when it moves away from the center towards + 100 or -100 (that is, if it goes towards the extremes) we can define it UP when it returns to the center we can define it DOWN
basically we calculate the absolute value of the stick position
then we choose whether to apply the delay when it increases or when it decreases
extending the discussion, one could define a resting position and when one moves away from it one condition applies while when one approaches it another applies.
currently it is considered UP if it goes from -100 to 100 and down if it goes from 100 to -100

I assume though that the whole thing is a bit complicated to make

nikpoz, 2024-06-04:
Yes , this is what I meant....Apologies if my description was poor..

MicheleVilla, 2024-06-04:
By dividing the mixer in two
one with function >0 and slowdown on UP
the other with function < 0 and slowdown DOWN
maybe you get the same thing ?

3djc, 2024-06-05:
I wonder if we could have this feature with an 'center based' checkbox, that would determine up/down related to center

pagrey, 2024-06-05:
This is to replace one logical switch and a single extra mixer line. The terms increase and decrease with absolute value make sense to me since that's how the logical switches are defined. For most people up and down makes sense but then center based doesn't make sense.

It seems like a simple example in the guide would work. I'd hate to see the UI become more and more complicated because people don't want to learn how to do two lines of setup.

chofchop, 2024-06-23:
I would like to join this discussion at a later date.

I also agree with [nikpoz](https://github.com/nikpoz) and [MicheleVilla](https://github.com/MicheleVilla).

Generally, when |x| increases, it is set as "servo turn speed", and when |x| decreases (when approaching neutral), it is set as "servo return speed".

To run this with EdgeTX 2.10.1, I split the mixer into x>0 and x<0 and set the turn speed to 3 seconds and return speed to 0 seconds.
The result is as shown in the video, there is a delay when crossing neutral, It is of no use.

https://github.com/EdgeTX/edgetx/assets/154777776/a4957d07-7f7e-44f3-b101-dcc4b9411c33

>[pagrey](https://github.com/pagrey) It seems like a simple example in the guide would work. I'd hate to see the UI become more and more complicated because people don't want to learn how to do two lines of setup.

I basically agree with you, but isn't this an EdgeTX issue? If you know of a better "two lines of setup" please let me know.

>[pagrey](https://github.com/pagrey) For most people up and down makes sense but then center based doesn't make sense.

I don't think that's the case. For center-based servo channels such as rudder and aileron, setting the turn and return speed is an extremely basic and important function.

pagrey, 2024-06-23:
This is a primary control. If you want a different speed then just move the control at the speed you want. Delay and speed are for outputs that don't have direct control over speed like switches for flaps or for outputs that you just want to set quickly and get back to primary control. I'd be interested to hear a use case where you can't just move the control at the desired rate? The more I think about it the less I understand what the possible use for this is.

inventor7777, 2024-06-23:
@chofchop while unrelated, how did you make the bezels on your MT12 screen black? Did you create your own sticker, or did you buy it? Or is it just an OLED replacement?

I’m saying that because I know that when the standard screen is set to inverted, the bezels stay white.

pfeerick, 2024-06-23:
I this one of the issues here is may be that is something that appears to be more common in surface transmitters, so is a foreign concept to us. Recent got a Flysky NB4 so will be coming to grips with some of the surface specific stuff myself now ;)

@inventor7777 That looks like an OLED replacement display, has the sharp whites and blacks that you can't have if you run normal LCD inverted ;) You can also make out the frame of the display at the botton ;)

pagrey, 2024-06-23:
Details of an OLED conversion are [here](https://www.rcgroups.com/forums/showpost.php?p=52372431&postcount=674) and [here](https://github.com/DongkwanYi/OLEDBOXER).

chofchop, 2024-06-24:
> This is a primary control. If you want a different speed then just move the control at the speed you want. 

As you say. I think I should too. But only some experts can do it, and many people need electronic assistance.

> I'd be interested to hear a use case where you can't just move the control at the desired rate?

Since [nikpoz](https://github.com/nikpoz) started talking about MT12, I will give an example of a steering servo for an RC car.
For optimum cornering in a racing car, not only the timing and amount of steering, but also the speed of steering are important. Top drivers do this with their own hands, but it is difficult for the average person, so they rely on the radio's "assist function." It is similar to using EXPO or custom curves.

pagrey, 2024-06-24:
This is going to do nothing to help your driving, you might look into a gyro. Delayed or slow steering for racing, that's a new one.

chofchop, 2024-06-24:
It's completely different from a gyro.
I'm sorry if I'm wrong, but I think you have no experience with racing cars. Or are you a super expert driver?
In any case, the concept of servo speed has been around for decades, and it is true that there are many people who need it.

philmoz, 2024-06-24:
Define 2 curves, one for 'up' and one for 'down':
[image]

Define 2 inputs for the desired channel, one uses the 'up' curve, the other the 'down' curve (AilP and AilN here):
[image]

Define two mix lines, one for the 'up' input and one for the 'down' input, mix lines are added:
[image]

Set slow up and slow down on each mix line as needed.

chofchop, 2024-06-24:
[philmoz](https://github.com/philmoz), Thank you, I did what you said and got the desired behavior.
I don't have time right now, so I will report in detail later.

inventor7777, 2024-06-24:
@pfeerick and @pagrey thanks for the links and info.

philmoz, 2024-06-25:
> For example, set the turn speed to 0sec. and the return speed to 3sec., it will look like the following video.

Is that realistic - I thought you wanted to have a slow turn and return?

On my MT12, if I set both slow up and slow down, then it behaves as expected (as far as I can tell).

Also keep in min the slow speeds are for full travel from -100 to +100.
Since you are setting slow speeds for only half the range you may need to adjust your values accordingly.

inventor7777, 2024-06-25:
This makes much more sense now that I’ve reread it with that in mind. I myself would also like this feature.

philmoz, 2024-06-25:
> What I'm trying to say is not whether there is a practical problem, but that I can't find a way to achieve the expected behavior.

I must be missing something - what expected behaviour does my solution not meet?

philmoz, 2024-06-25:
The curves solution works when speed up >= speed down; but not the other way around.

Here is a solution using logical switches and mix lines that works regardless of how speed up and down are set (at least it does on my MT12).

Create a logical switch to detect when the output channel is positive - the important part is using the output channel not the input:
[image]

Create two mix lines, both switched with the logical switch. The second one should invert the switch. The first line is positive steering, the second is negative. The second mix line is set to replace mode.
[image]

Set the speed up and speed down on the two mix lines.

This will then use the positive speed up/down settings until the output channel goes negative and vice versa.

MicheleVilla, 2024-06-26:
splitting Mixer in two line with option Curve set to Func  X>0 and Func  X<0  we obtain the same behaviour
I believe, however, that there is a need to define the underlying operating principle and that is to distinguish between types of controls that have a single stroke and those that have a central position

philmoz, 2024-06-26:
> splitting Mixer in two line with option Curve set to Func X>0 and Func X<0 we obtain the same behaviour I believe

I tried this and it does not work.

chofchop, 2024-06-26:
@philmoz That's excellent. I had no idea it was possible to achieve this behavior.
Once again, I am impressed by EdgeTX's flexibility and your intellect.

However, it would be cruel to ask the average person to get there on their own, so is there a better way to make it easier for anyone to learn about this method?

MicheleVilla, 2024-06-26:
> > splitting Mixer in two line with option Curve set to Func X>0 and Func X<0 we obtain the same behaviour I believe
> I tried this and it does not work.

https://github.com/EdgeTX/edgetx/assets/6508436/55c09fd1-d306-43bb-b051-efdd41521e7f

MicheleVilla, 2024-06-26:
[image]

chofchop, 2024-06-26:
@MicheleVilla ,Your settings are now 5 sec. for up and 0 sec. for down. That's no problem.
Try setting up to 0 sec. and down to 5 sec,, and try moving the stick left and right as if straddling the center. I think you can observe different behavior depending on the settings.

MicheleVilla, 2024-06-26:
> @MicheleVilla ,Your settings are now 5 sec. for up and 0 sec. for down. That's no problem. Try setting up to 0 sec. and down to 5 sec,, and try moving the stick left and right as if straddling the center. I think you can observe different behavior depending on the settings.

You are right!
Is strange behaviour
Is a problem or is normal?
perhaps one more reason to better define this feature

chofchop, 2024-06-26:
If you set the down to 5 seconds, the servo (output) will still remain on the left when you quickly move the stick to the right.
Even though the servo (output) is still on the left, the mixer line's "right" settings are applied.

So as @philmoz said, you need to refer to the "output" and switch the mixer line.

MicheleVilla, 2024-06-26:
In @philmoz's solution the two mixers are disabled when output=0
In the case of slowdown set to return to the center when you instantly switch from the stick to -100 to 100 the movement is slowed down to 0 and immediately switches to +100
I wonder if from a practical point of view it makes sense

chofchop, 2024-06-26:
> In @philmoz's solution the two mixers are disabled when output=0

That won't happen.
L01 is defined to be true when 0>x. In other words !L01 is x>=0.
If x = 0, the second mixer line will be active.
Also, I think there is a meaning to replace mode, but I don't fully understand it either.

inventor7777, 2024-08-20:
Any updates on this feature?

jasc76, 2024-09-11:
Isn't the requested function more or less a filter with really low cutoff and depending if the direction is pos or neg ?