EdgeTX/edgetx #5121: Trims T1 and T2 don’t show up as hardware even when set to 3POS.

inventor7777, 2024-06-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I’ve set trims T1, T2, T3, and T4 to 3POS switches. However, while T3, T4, and T5 appear as options for LS, LF, and Mixes, T1 and T2 don’t appear.

### Expected Behavior

I expect all trims to appear (T1, T2, T3, T4, and T5).

### Steps To Reproduce

Under Drive Modes, set all trims to 3POS. Try using T1 or T2 in any LS, LF, or Mix.

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

The “beep” audio turns off on set trims, so I know I did it right. The radio is a RadioMaster MT12 4-in-1 running the latest 6/3/24 Nightly.

philmoz, 2024-06-03:
They will probably be called ST and TH, not T1 and T2.

pfeerick, 2024-06-03:
They are.

inventor7777, 2024-06-04:
@philmoz @pfeerick you guys are correct. Is there any chance those could be renamed in the future for better usability? (To T1 and T2) or at least have an option to do so?

elecpower, 2024-06-05:
> @philmoz @pfeerick you guys are correct. Is there any chance those could be renamed in the future for better usability? (To T1 and T2) or at least have an option to do so?

I support the enhancement to configure trims like sticks, pots and switches. especially as support for surface radios grows and the other uses to which radios are put.

inventor7777, 2024-06-05:
> I support the enhancement to configure trims like sticks, pots and switches. especially as support for surface radios grows and the other uses to which radios are put.

Exactly! Especially as many surface people don’t need more than two trims anyway.