EdgeTX/edgetx #5126: Models transfer looses settings moving from OpenTX to EdgeTX

jtapernoux, 2024-06-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I have migrated my Jumper T16 from OpenTX to Edge TX.  I converted my model.otx file using EdgeTX Companion to a model.etx file, and loaded it on the transmitter.  Models are there now, but several settings in the Logical Switches and Special Functions pages are either missing or have different values.  Is this expected, and I have to manually make corrections, or have I followed a procedure incorrectly?
Thanks,
John

### Expected Behavior

I did not expect to find missing or altered values in my model files.

### Steps To Reproduce

load model.otx into edgetx companion, then save as model.etx, then load model.etx to transmitter

### Version

2.10.1

### Transmitter

Jumper T16

### Operating System (OS)

Windows

### OS Version

Windows 11

### Anything else?

Tried the same thing using a Radioking TX18S, with the same results.

pfeerick, 2024-06-05:
Can you upload the OTX file so we can see what is going on? That is basically the expected process... i.e. load OTX file with EdgeTX Companion, let Companion do it's conversion changes, etc.  Generally, Companion should be able to convert everything over, but sometimes not everything is able to be converted over automatically... don't think there is anything in logical switches that should be a problem there, so probably a bug.

jtapernoux, 2024-06-05:
Thanks Peter, I’ll upload the file.  I’ll point out a couple settings that converted wrong.JohnSent from my iPad

jtapernoux, 2024-06-05:
Not finding where I can upload the OTX file...

jtapernoux, 2024-06-05:
ETX File "Models OpenTX 24-06-03" model "Spitfire":   L01 missing "And Switch: 6P_6", L02 missing "V2: 6P_5
                                                                                                                 SF1 and SF2 completely blank, also involving 6P_6 and 6P_5
ETX File "Models EdgeTX May, Active" model "Spitfire":  Special Functions "Repeat Value" changed from "No Repeat" to "Repeat 1S"

3djc, 2024-06-05:
Try transfering using EdgeTX companion 2.9.4 first, save to .etx, open then etx with EdgeTX companion 2.10, it should be better

pfeerick, 2024-06-05:
With a zip file extension, you should be able to upload it using this option ;)

[image]

elecpower, 2024-06-05:
I would appreciate getting a copy of the otx to see what I missed.

jtapernoux, 2024-06-05:
[Models EdgeTX May, Active.zip](https://github.com/user-attachments/files/15592405/Models.EdgeTX.May.Active.zip)

jtapernoux, 2024-06-05:
Hi,
Using Edge Companion 2.9 to convert OTX to ETX, then Edge Companion 2.10 to load radio worked!  I had to enable the 6 position switches in the hardware calibration menu too.
Thanks!!!

elecpower, 2024-06-05:
Thank you for uploading the files. Checked and confirmed the 6 position switch does not convert from otx.

elecpower, 2024-06-12:
What was the sequence to change the No Repeat to Repeat 1S as it is not altered in the otx conversion?
PS fix for multi position switch awaiting approval and merging

PR #5169 (2024-06-12): fix(cpn): binary import of multi position switches
Fixes #5126
closed 2024-06-12 by pfeerick