EdgeTX/edgetx #5128: Using the " symbol in model name causing SD card error when enabling SD logging.

Woofie1010, 2024-06-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

While using a model name: Laser V3 67" I receive an SD card error message when logging the flight. After removing the " symbol the SD card error did not happen.

### Expected Behavior

If the symbol is not aloud it should not be a possibility. 

### Steps To Reproduce

1.Put the symbol " in the model name. 2. Enable SD Loggin in special functions

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

pfeerick, 2024-06-06:
While it can't be entered from the colorlcd radio keyboard, I think `"` was supposed to be an allowed character. #4670 verifies that only valid characters are permitted in model and label names when entered via Companion, so we either need to drop it from the list of valid characters, or hunt down why this happening.

philmoz, 2024-06-06:
Double quote is an allowed character in the model name on the radio.
It is not an allowed character in a filename - the logs code is trying to create a file using the model name so the file create fails.

Replacing the double quote with an underscore in the log filename fixes it.

pfeerick, 2024-06-06:
>  the logs code is trying to create a file using the model name so the file create fails.

ouch... it needs to be restricted to only valid file-system characters then!!

closed 2024-06-06 by pfeerick

Woofie1010, 2024-06-06:
Peter, Thanks for the update. I will not be using the " symbol in a model name.
Dave

Woofie1010, 2024-06-06:
Thanks for the update. I will not be using the " symbol in a model name.
Dave

pfeerick, 2024-06-07:
No problem. It shouldn't be an issue as of 2.10.2 or later though.

Woofie1010, 2024-06-07:
Peter,
Big thanks to you and the team for your work. 
I love this stuff.Dave