EdgeTX/edgetx #5137: Companion 2.10.1 - Profile conversion broken

RC-SOAR, 2024-06-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

Attempting to import my DLG6S template using Companion 2.10.1. The template was designed for X9D, importing into Zorro profile.  It worked fine in 2.9, now the process is broken.
Problems:
1. The Radio settings > Hardware tab is overwritten with X9D switch definitions which are not relevant to the Zorro.
2. The conversion messages contain several irrelevant and confusing messages as a result of (1) above.

[image]
[image]

### Expected Behavior

Works correctly in Companion 2.9 - switch definitions are not altered, and the single error message is meaningful.
[image]
[image]

### Steps To Reproduce

Import the DLG6S template (attached) into Companion 2.10.1
[DLG6S_200.otx.txt](https://github.com/user-attachments/files/15748651/DLG6S_200.otx.txt)

### Version

2.10.1

### Transmitter

RadioMaster Zorro

### Operating System (OS)

Windows

### Anything else?

Critical issue for my sailplane templates.

elecpower, 2024-06-10:
Many thanks for the file. I will have a look.

elecpower, 2024-06-10:
Confirmed. On my TODO list

PR #5151 (2024-06-12): fix(cpn): profile conversion of models and settings
Fixes #5137

Summary of changes:
- fix use profile default hardware switch type when file switch type not supported eg convert 3 Position to 2 Position
- fix hardware switch conversion to type (was reporting from)
- add 'unsupported' conversion type
- use 'unsupported' for hardware not existing in destination radio instead of 'invalid'
- when values 'invalid' also report conversion applied (future merge invalid and conversion rows - separate in this PR for expediency)

[image]
closed 2024-06-12 by pfeerick