EdgeTX/edgetx #5139: Trainer SBUS does not work on TARANIS QX7-Access

bug400, 2024-06-08:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have a SBUS source connected to the external module bay. After upgrading to 2.10.1 from 2.9.3 there is no trainer input any more. If I switch to a model with "master/SBUS" mode I get neither any haptic feedback nor any trainer input.

### Expected Behavior

Power on external module (haptic feedback). Get trainer input from SBUS.

### Steps To Reproduce

Change to a model with "master/SBUS" mode. Watch for haptic feedback, and TRx values.

### Version

2.10.1

### Transmitter

FrSky Q X7 / Q X7S (ACCESS)

### Operating System (OS)

Linux

### OS Version

Debian Bookworm

### Anything else?

This is a regression error because of the refactoring of the trainer driver code in 2.10. I am working on a fix.

mha1, 2024-06-09:
Have you tried using the S.Port pin for SBUS in?

pfeerick, 2024-06-09:
But *why* should 2.10 be different to 2.9? I think that is the key detail
here, not which pin is used. ;) Especially if the linked PR resolves this.

mha1, 2024-06-10:
Because of #4668

raphaelcoeffic, 2024-06-10:
> Because of #4668

Right, the `#if defined()` where not necessary in the first place (see #5149)

commit 21fbf10f (2024-06-10): fix: check both SBUS trainer ports regardless of TRAINER_MODULE_SBUS_USART

Fixes #5139
PR #5149 (2024-06-10): fix: check both SBUS trainer ports regardless of TRAINER_MODULE_SBUS_USART
Fixes #5139

Summary of changes:
- removed `#if defined(TRAINER_MODULE_SBUS_USART)` and just check presence of potential ports instead.
closed 2024-06-10 by pfeerick