EdgeTX/edgetx #5150: Cannot navigate to enable checkbox in special functions

wumfi, 2024-06-10:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

Using a Jumper T-Lite v2, running v2.10.0-rc3.

On the special functions screen for a model, there appears to be no way of reaching the enable checkbox. Is this a bug, or user error?

### Expected Behavior

Ability to enable/disable a special function

### Steps To Reproduce

1. Create new SF
2. Attempt to enable

### Version

2.10.0

### Transmitter

Jumper T-Lite

### Operating System (OS)

Linux

### OS Version

Not relevant

3djc, 2024-06-10:
Maybe you can try latest release version , like 2.10.1 ?

wumfi, 2024-06-10:
I shall update and report back. Thanks.

pfeerick, 2024-06-10:
I can't reproduce this with 2.10.0. 2.10.0-RC4 (and thus also in subsequent release version of 2.10) most likely fixed this via #4967

3djc, 2024-06-11:
Me asking wasn't a shot in the dark ;)

wumfi, 2024-06-12:
Confirmed fixed in 2.10.1. Thank you!

closed 2024-06-12 by 3djc