EdgeTX/edgetx #5153: Taranis X-Lite Pro ISRM not selectable on TX

druckgott, 2024-06-10:
If I want to select ISRM on the TX it´s not possible.
On the TX I only get all other ones, but on ISRM.
If I select in in Compainion its possible an If I take a look on the TX is selected if I write the config.
Then I go again to the TX select something als and the ISRM dissapears again. Then I have to reactivate it again over Companion.

[image]

pfeerick, 2024-06-10:
This was opened in the wrong place.

What version of EdgeTX are you running? And why are you trying to change it in the hardware settings? 

I can reproduce this in simulator for 2.10.0 FWIW.

druckgott, 2024-06-11:
I use 2.10.0

PR #5236 (2024-07-01): fix: supported internal module for monochrome radios w/ ISRM
Fixes #5234
Fixes #5153

pfeerick: I am going to tentatively say this resolves #1942 also as this gem has been present since 2.6, so could have been responsible since it was a 2.7 issue. 

Summary of changes:
- use `DEFAULT_INTERNAL_MODULE == ISRM_PXX2` as a better heuristic to determine the supported modules.

Note: it seems X9lite and X9liteS are impacted as well.
closed 2024-07-01 by pfeerick