EdgeTX/edgetx #5157: backup file path ignored in 2.10.1

19Roger53, 2024-06-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Companion

### Current Behavior

I just installed 2.10.1 and entered a file path for back ups but my chosen path is ignored and replaced by the wrong folder

### Expected Behavior

I expect the file path to be recognized 

### Steps To Reproduce

I have been using Edge tx companion 2.8.1 since its inception
1. installed Edgetx  companion 2.10.1
 2. in settings I selected a file path for my models backup folder
3. went to file dialog box, chose 'open' and hit enter
4. it then opened to the wrong folder

I re-entered the folder that i wanted multiple times but it kept going to the wrong folder

i also deleted 2.10 and re-installed but still have the same problem

Thank you for your help

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

### OS Version

windows 10

### Anything else?

[image]
[image]
[image]
[image]

19Roger53, 2024-06-11:
correction: the OS on that PC is Windows 8.1

elecpower, 2024-06-11:
That is the folder location for each radio backups. It is used by the Read/Write menu items Backup Radio tot File and Write Backup to Radio both of which are currently disabled. 
Also used when the checkbox below the folder location is checked for the purpose stated.
Companion should save and open the last used settings folder but there is an issue #4813 identifying that this is not working.

pfeerick, 2024-06-12:
I think there are two separate issues here... 

The first time you run through the steps given
```
1. installed Edgetx companion 2.10.1
2. in settings I selected a file path for my models backup folder
3. went to file dialog box, chose 'open' and hit enter
4. it then opened to the wrong folder
```

I would'n't really expect it to open the path set in step 2 in step 3. `File -> Open` for is not specifically for opening the backup folder. Don't confuse `Backup folder` with (a nonexistent) `Default Open/Save Folder` option. 

That though would be nice to have, both at Radio Profile and Application Settings level... i.e. another field at the radio profile level that if blank reverts to Application Settings level setting, otherwise always uses to the configured path for open/save dialogs. At the application settings tab level, it would have similar to the "automatic backup folder" layout... a field that is blank by default, which is used as the global open/save path if set, and the checkbox would be to disable the field, and instead use last visited folder.

19Roger53, 2024-06-13:
An extra bit of information to hopefully clarify my understanding: In the past, using ETx 2.8 and several years of OpenTx my steps to get into my saved files were as I outlined: In Open or EdgeTx, hit File, then Open, which opened to the named Backup folder (from the settings page) with my saved model and radio settings files. After discovering that doesn't happen in 2.10.1 i'm thinking i have been blessed all these years for it to have been working that way.

3djc, 2024-06-13:
Clearly I would not have it operate this way. Remember last used directory yes, open backup, nop please

chuckstarck, 2025-11-09:
Same problem with 2.11.3 companion. Start companion, open backup file location is correct. First time after starting companion, after reading from radio (D:/) if you try to save file it is the the right root but not in the right folder. If you have C:/models/horus for the backup location, it defaults to C:/ (example). If you open files from the backup before reading from radio, then it's in the right folder.
OTX did what settings were.