EdgeTX/edgetx #5158: Why won't this music play?

inventor7777, 2024-06-11:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have successfully created custom sound files in the past (following the instructions on manual.edgetx.org (
File Name: 123456.wav (up to 6 characters plus .wav)
Sample Rate: 32 kHz (or 16 Khz, 8Khz)
Bits / Sample: 16 (or 8)
Tracks: 1, mono
Compression Codec: PCM)

However, I can't get my most recent sound to play "xp2.wav" (which is, incidentally. a slightly modified Windows XP boot sound).

Why is this? I have the export settings as follows: (shows in screenshot too) 32000Hz, 16 bit signed PCM, mono, 3Ch name. This worked on the "hello.wav" file also linked below. 
[image]

Affected file:
[xp2.wav.zip](https://github.com/user-attachments/files/15793035/xp2.wav.zip)

Non-affected custom file:
[hello.wav.zip](https://github.com/user-attachments/files/15793046/hello.wav.zip)

### Expected Behavior

I expected it to play.

### Steps To Reproduce

Export file as a WAV 16 bit PCM 32000hz (I tried random hz options on a known working sound file and 22050 and similar worked okay and played back) etc with Audacity and it doesn't play. (My other sound file was also made with Audacity)

### Version

Nightly (Please give date/commit below)

### Transmitter

RadioMaster MT12

### Operating System (OS)

macOS

### Anything else?

I'm running 2.11 nightly, 6/10/24.

philmoz, 2024-06-11:
The xp2.wav file you uploaded is 22050Hz.
EdgeTx only supports 32000Hz or something that evenly divides 32000 (e..g 16000Hz, 8000Hz, 4000Hz).

pfeerick, 2024-06-11:
As Phil said:
[image]

In Audacity, you need to ensure the *Project* sample rate is set to 32000Hz or other valid sample rate, as that will be used when you export the audio, not the individual sample rates of the clips. In older versions, it used to be right on the bottom toolbar, like here... 

[image]

On newer versions of Audacity, confusingly I think it is now under Audio Setup (button on toolbar) -> Audio Settings.

closed 2024-06-11 by pfeerick

inventor7777, 2024-06-11:
Ohhhhhhh really? That makes sense. It must have just been Audacity's weird interface because I don't see any "project rate" near the bottom of my screen. (I thought the clip sample rate would work) I'll look around for the Project Rate under Audio Setup :) 

Thanks for the help!