EdgeTX/edgetx #5163: Able to disable Touch Screen

ror1948, 2024-06-12:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

We have RC members who are in their senior years and they tend to be touching the touch screen on Tx16s unknowingly changing the setup.
They do not know how to setup Tx16s and it is done by someone else.

### Describe the solution you'd like

A sequence of button pushes to enable or disable touch screen 

### Describe alternatives you've considered

1) A sequence of button pushes to enable or disable setup changes on touch screen. This would allow the basic function of the touch screen

pfeerick, 2024-06-12:
Using the "No Touch" special function to disable the touch screen depending on the state of a switch or even some logic condition  isn't sufficient in this case?

3djc, 2024-06-12:
You could easily have a touch disabled by default, and enable it with a combination of t5/t6 position for example, just a logical switch and a special function

ror1948, 2024-06-12:
sorry i did not know there was a no touch option...thanks

closed 2024-06-12 by ror1948

ror1948, 2024-06-12:
ok but no logic switches available under SYS that i can see. running 2.10.1how do you disable it by default?thanks

pfeerick, 2024-06-12:
You can't use logical switches in global functions as they are model
specific. If you wanted to use a logical switch, you would have to add this
as a special function for each existing model. And maybe include it in a
new model template.