EdgeTX/edgetx #5165: Radiomaster tx16s edgetx Frozen screen when model select button pressed

poobah88, 2024-06-12:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

When running edge TX 2.8 or greater on a radiomaster tx16s the screen freezes when you press the model select button. I think I've found a work around by going into global setting and deselecting, or turning off, the global settings.
Anyone else having this problem. I've returned a few tx16s mk2 and they've all done it at various points.

### Expected Behavior

When you enter the model select button, it freezes and nothing works. You have to remove battery or hold down power button for like 30 seconds.

### Steps To Reproduce

Every time you press the model select button. Press the scroll wheel down and then choose the model select and press the scroll wheel down again and it freezes

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-06-12:
Are you able to replicate using another sd card with default content? (Not your model, just default)

Zaquron, 2024-11-06:
i have the same problem but it is not consistent.
Sometimes it works flawlessly, other times it takes 3 "unplugs" to get to the model select screen. 
I will try a different SD card tomorrow.

pfeerick, 2025-02-14:
Any progress / updates on this?

Zaquron, 2025-02-15:
Sorry i forgot to update. 
I manaeged to solve the issue with properly resizing my .png files

closed 2025-02-15 by pfeerick