EdgeTX/edgetx #5189: AUX1 powered at startup with AUX1 power set to off

mha1, 2024-06-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

At the instance of pressing the power button AUX1 is powered even if AUX1 port power is set to off. AUX1 stays powered for the entire life cycle unless you set AUX1 port power to on and off again. 

Example: receiver connected and powered with port power set to off

[image]

### Expected Behavior

AUX not powered if port power set to off.

### Steps To Reproduce

1. make sure AUX1 port power is set to off
2. turn on radio
3. measure voltage between AUX1 GND and VCC pins

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-06-16:
Not happening here: https://photos.app.goo.gl/8aHWFDqhVtUyzKsw8

philmoz, 2024-06-16:
If the port is set to OFF then the power state is not updated in serialInit() (serial.cpp).

3djc, 2024-06-16:
Ok, port role need to be OFF and port power, understood

PR #5190 (2024-06-16): fix: serial power port handling
Fixes power been set when serial mode set to OFF and serial port power also set to OFF

Also fixes ON glitch at power on.

Fixes #5189
closed 2024-06-16 by pfeerick