EdgeTX/edgetx #5191: Enable/Disable Logging to SD Card from LUA

derelict, 2024-06-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

no

### Describe the solution you'd like

add a LUA Function to enable/disable Logging to SD Card

### Describe alternatives you've considered

none

### Additional context

i have tried to do it myself in the code ... but i have no knowhow with C++ ... i'm sure for a DEV it would be easy to do. I have managed to create a LUA Function that turns logging on or off by modifying initLoggingTimer() logsWrite() with a Boolean value ... but the status was not reflected on the UI.

mha1, 2024-07-05:
Ich denke weder Handyakku, noch Powerbank werden lange im Kasten überleben.

mha1, 2024-08-17:
You can do this with LUA itself. Attached find a LUA function to enable/disable a defined SF SD Logs plus a little init routine that needs to be run once in order for the SD Logs enable/disable function to work.

For demo purposes the LUA function is embedded in a little function script to show how it is used. Just rip out `function initSDLogEnable()` and `function SDLogsEnable(enable)` and use it in your code.

Ho does it work? ini`tSDLogEnable()` searches for a defined SF SD Logs in the current selected model setup. If it finds one it memorizes its definition. `SDLogsEnable(0)` or `SDLogsEnable(1)` uses the stored SF data and overrides the enable status. To see what the demo does install Logs01.lua as function script and define a special function SD Logs. The function script will enable/disable the logs every 5s. The UI reflects the status accordingly.

derelict, 2024-08-19:
> You can do this with LUA itself. Attached find a LUA function to enable/disable a defined SF SD Logs plus a little init routine that needs to be run once in order for the SD Logs enable/disable function to work.

will this work ... even if the transmitter does not have an SF for logs defined/set ?

mha1, 2024-08-19:
currently no but init can be extended with a few lines of code to automatically create an SF in case it doesn't find one.

derelict, 2024-08-23:
Cool. I will give it a try in my widget (currently using "radio made SF"). Thanks for eh Hint/Input !

pfeerick, 2024-08-24:
Gonna close this as resolved as like how https://github.com/EdgeTX/edgetx/issues/5191#issuecomment-2294957320 is doing it is probably the only way this can really be reasonably be done, given this is a SF you need to define in the first place, and can be any line, any trigger, any state. So the API to do what you want can't be as 'simple' as a `sdlogging(true/false)` function... i.e. as you have multiple SD lines actually work, to configure logging at different times/rates, etc. It really does need to be "find the SD Logging SF of interest, enable or disable it" approach.

closed 2024-08-24 by pfeerick

derelict, 2024-09-28:
> Here's the version that'll create a SF SD Logs in the first free SF slot

Could it be, that this is not working when called/used in a widget script ?

If i execute your script manually it works ... but once "embedded" in a widget script it doesn't work anymore.

`local test = model.getCustomFunction(SDlogsSFIndex)		-- get SF at index i

test.active = 1
test.value = 30

model.setCustomFunction(SDlogsSFIndex, test)

local test = model.getCustomFunction(SDlogsSFIndex)		-- get SF at index i

print("SFDEBG: CF Debug " .. test.switch )
print("SFDEBG: CF Debug " .. test.func )
print("SFDEBG: CF Debug " .. test.value )
print("SFDEBG: CF Debug " .. test.active )
`

returns the correct values ... but nothing is changed/visible at the SF Screen ... and logging is not on ... even though "test.active" says "1"

`
SFDEBG: CF Debug 233
SFDEBG: CF Debug 18
SFDEBG: CF Debug 30
SFDEBG: CF Debug 1
`

mha1, 2024-09-29:
I don't know but can't think of a reason why it wouldn't work in a widget. Please attach a minimal version of your widget demonstrating the problem.