EdgeTX/edgetx #5215: Trainer Connection via MULTI instable

ParkerEde, 2024-06-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### What part of EdgeTX is the focus of this bug?

Transmitter firmware

### Current Behavior

I have two Radiomaster TX16S with 4in1 MPM and an additional external iRange+ 4in1MPM. I would like to use the teacher/student function. The teacher radio uses the internal MPM for connection to the model and the external MPM as RX for the trainer signal. I have configured everything so far and it basically works. Unfortunately, the connection is extremely unstable and I keep getting continuously the message Student signal lost/reestablished. I have selected Frsky X2 EU/LBT for the trainer connection.
Is this a problem with EdgeTX 2.10.1 or rather with the MPM with the current firmware 1.3.4.0?

### Expected Behavior

MULTI Trainersignal should be stable

### Steps To Reproduce

Trainer Radio:
1. Use ext. 4in1 MPM as Frsky X2 RX
2. set Trainer Signal  to MULTI
Student Radio:
1. Use int. 4in1MPM with Frsky X2 EU/LBT

I also used a Horus X10 Express as a student radio and the same problem occurs.

### Version

2.10.1

### Transmitter

RadioMaster TX16S / TX16SMK2

3djc, 2024-06-24:
If your plane is using frsky protocol, you should try something else, like spektrum, for the trainer/master connection.

3djc, 2024-06-24:
There are known RF issues with certain protocols when trying to use the same in super close proximity

ParkerEde, 2024-06-24:
okay. Thank you. I will check.

ParkerEde, 2024-07-31:
with DSM for Trainer it's okay

closed 2024-07-31 by ParkerEde